openhwfoundation/cva6 #976: Increase base frequency of CVA6 core

0ena, 2022-10-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I would like to increase the base frequency of the CVA6 core from 50MHz to a slightly higher frequency, let's say 53MHz.

For that, I changed the `CLKOUT1` in the Xilinx clock generator to output 53 MHz.
Indeed the new `CLKOUT1` frequency is now 52.980MHz (18.875 ns period), but I get big SETUP violations on the `dmi_jtag` module, between clocks `tck` and `clk_out1`.

Bellow follows an example of those paths:

```
Timing Report

Slack (VIOLATED) :        -5.130ns  (required time - arrival time)
  Source:                 i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_src/data_src_q_reg[data][16]/C
                            (rising edge-triggered cell FDCE clocked by tck  {rise@0.000ns fall@50.000ns period=100.000ns})
  Destination:            i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_dst/data_dst_q_reg[data][16]/D
                            (rising edge-triggered cell FDCE clocked by clk_out1_xlnx_clk_gen  {rise@0.000ns fall@9.438ns period=18.875ns})
  Path Group:             clk_out1_xlnx_clk_gen
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            0.125ns  (clk_out1_xlnx_clk_gen rise@10400.125ns - tck rise@10400.000ns)
  Data Path Delay:        0.590ns  (logic 0.223ns (37.829%)  route 0.367ns (62.171%))
  Logic Levels:           0
  Clock Path Skew:        -4.060ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    1.609ns = ( 10401.734 - 10400.125 )
    Source Clock Delay      (SCD):    5.669ns = ( 10405.669 - 10400.000 )
    Clock Pessimism Removal (CPR):    0.000ns
  Clock Uncertainty:      0.595ns  ((TSJ^2 + TIJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Total Input Jitter      (TIJ):    1.000ns
    Discrete Jitter          (DJ):    0.128ns
    Phase Error              (PE):    0.090ns
  Clock Domain Crossing:  Inter clock paths are considered valid unless explicitly excluded by timing constraints such as set_clock_groups or set_false_path.

    Location             Delay type                Incr(ns)  Path(ns)    Netlist Resource(s)
  -------------------------------------------------------------------    -------------------
                         (clock tck rise edge)    10400.000 10400.000 r
    AD27                                              0.000 10400.000 r  tck (IN)
                         net (fo=0)                   0.000 10400.000    tck
    AD27                 IBUF (Prop_ibuf_I_O)         1.518 10401.518 r  tck_IBUF_inst/O
                         net (fo=1, routed)           2.312 10403.829    tck_IBUF
    BUFGCTRL_X0Y7        BUFG (Prop_bufg_I_O)         0.093 10403.922 r  tck_IBUF_BUFG_inst/O
                         net (fo=258, routed)         1.746 10405.668    i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_src/tck_IBUF_BUFG
    SLICE_X99Y35         FDCE                                         r  i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_src/data_src_q_reg[data][16]/C
  -------------------------------------------------------------------    -------------------
    SLICE_X99Y35         FDCE (Prop_fdce_C_Q)         0.223 10405.891 r  i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_src/data_src_q_reg[data][16]/Q
                         net (fo=1, routed)           0.367 10406.257    i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_dst/data_dst_q_reg[data][31]_2[16]
    SLICE_X95Y35         FDCE                                         r  i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_dst/data_dst_q_reg[data][16]/D
  -------------------------------------------------------------------    -------------------

                         (clock clk_out1_xlnx_clk_gen rise edge)
                                                  10400.125 10400.125 r
    BUFGCTRL_X0Y1        BUFG                         0.000 10400.125 r  i_ddr/u_xlnx_mig_7_ddr3_mig/u_ddr3_infrastructure/u_bufg_clkdiv0/O
                         net (fo=11404, routed)       1.776 10401.901    i_xlnx_clk_gen/inst/clk_in1
    MMCME2_ADV_X1Y0      MMCME2_ADV (Prop_mmcme2_adv_CLKIN1_CLKOUT0)
                                                     -4.195 10397.706 r  i_xlnx_clk_gen/inst/mmcm_adv_inst/CLKOUT0
                         net (fo=1, routed)           2.336 10400.042    i_xlnx_clk_gen/inst/clk_out1_xlnx_clk_gen
    BUFGCTRL_X0Y0        BUFG (Prop_bufg_I_O)         0.083 10400.125 r  i_xlnx_clk_gen/inst/clkout1_buf/O
                         net (fo=38025, routed)       1.609 10401.734    i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_dst/clk_out1
    SLICE_X95Y35         FDCE                                         r  i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_dst/data_dst_q_reg[data][16]/C
                         clock pessimism              0.000 10401.734
                         clock uncertainty           -0.595 10401.139
    SLICE_X95Y35         FDCE (Setup_fdce_C_D)       -0.010 10401.129    i_dmi_jtag/i_dmi_cdc/i_cdc_req/i_dst/data_dst_q_reg[data][16]
  -------------------------------------------------------------------
                         required time                      10401.129
                         arrival time                       -10406.258
  -------------------------------------------------------------------
                         slack                                 -5.130

```

The negative slack seems pretty big considering the small change in the frequency.

I have seen in a previous question [here](https://github.com/openhwgroup/cva6/issues/389), that it is possible to increase the base frequency if I adjust the base frequency in the bootloader and the device tree.

Some questions:

1) which device tree should I adjust, the `cv64a6.dts` or the `ariane.dts`?
2) in which file do I change the base frequency of the bootloader?
3) are the above files used during Synthesis/Implementation and if so, are they sufficient to resolve my issue?

I am using a Genesys-2 board and Vivado v19.2.

Thank you in advance for your help and time!

Kind regards,
Nassos

0ena, 2022-10-13:
I managed to solve my SETUP violations by adjusting the maximum delay of the failing paths in the .xdc file of Ariane.
Implementation finishes now without any timing issues.

So to my understanding the remaining things are:
a) update the clock frequency of the core on the Device Tree.
b) update the clock frequency of the core in the bootloader.

Can someone please specify those files?
For the device tree it should either be the `cv64a6.dts`, the `ariane.dts` or both of them.
What about the bootloader?

Thank you in advance for your help and time.

Kind regards,
Nassos

MikeOpenHWGroup, 2023-02-07:
Hi @0ena, thanks for your interest in CVA6.  This is a detailed question that we are not positioned to resolve in a GitHub issue discussion.   I would encourage you to become directly involved in the OpenHW Group if you are interested in these types of improvements to our open-source IP.   Please reach out to me at mike@openhwgroup.org if you'd like to discuss this further.

If we do not hear from you in two weeks (by 2023-02-21), I will close this issue as a "Wont-Fix".

Thanks!

0ena, 2023-02-09:
Hi @MikeOpenHWGroup , thank you for reaching out.
I was able to solve this issue a while ago.
I increased the base clock frequency in both the `cv64a6.dts` and `ariane.dts` and relaxed the maximum delays in the `.xdc` file of the failing `dmi_jtag` paths to resolve the setup violations.
I have tested the core and it runs without issues so far.
I am closing this thread.

Sure, I'll shoot an e-mail in the following days! :-)

closed 2023-02-09 by 0ena