openhwfoundation/cva6 #977: [BUG] make veri_run in the README is missing supporting Makefile and testbenches

tianrui-wei, 2022-10-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The readme.md claims there's a core-only testbench with `make veri_run` [here](https://github.com/openhwgroup/cva6/blob/4c01614f83cd9dd91d115aa190bd4fc6a284b59e/README.md?plain=1#L140), but no Makefile or supporting testbench is available. It'd be great if someone could either update the instruction or add the supporting files

kaamakshee, 2022-12-10:
I am also facing the same issue. Unable to run a core test due to this. Please help!!

MikeOpenHWGroup, 2023-02-07:
Hi @ASintzoff.  According to "git blame", that comment about the core-only testbench was written by you.  Is this core-only testbench still supported?

ASintzoff, 2023-02-08:
Hello @MikeOpenHWGroup in this case, "git blame" is not your friend. It is not able to identify that you ;-) are the one having written this. Unfortunately in CVA6 repository, all commits of a pull request are squashed in only one commit. The original commit is part of PR #667.

Whatever, PR #1050 is available.

tianrui-wei, 2023-02-14:
Thanks for resolving this!

commit 01e2ad36 (2023-02-08): README.md: core only simulation done via core-v-verif (fix #977) (#1050)
closed 2023-02-14 by tianrui-wei