openhwfoundation/cva6 #980: [BUG] Compatibility with Verilator 5.0

flaviens, 2022-10-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi!

It looks like the current master branch is not verilatable with `Verilator 5.001 devel rev v4.228-185-g7b07cf912`

```
../corev_apu/tb/ariane_tb.cpp: In function ‘int main(int, char**)’:
../corev_apu/tb/ariane_tb.cpp:352:49: error: ‘class Variane_testharness’ has no member named ‘ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_tc_sram_wrapper__DOT__i_tc_sram__DOT__sram’
  352 |   memif.read(0x80000000, mem_size, (void *)top->ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_tc_sram_wrapper__DOT__i_tc_sram__DOT__sram);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [Variane_testharness.mk:68: ariane_tb.o] Error 1
```

MikeOpenHWGroup, 2023-02-07:
Hi @flaviens, thanks for your issue.  At this time we have not committed to support Verilator 5.  However, if you would like to submit a pull-request to resolve this issue we would certainly review it.

Jbalkind, 2023-02-18:
Given that verilator 5 supports timing and can generate its own C++ top level instead of requiring the user to write their own, I'm not sure it entirely makes sense to use the existing verilator setup when moving to version 5 (and thus I don't know if this issue is worth fixing). It may be better to unify with the top levels for the other simulators instead.

JeanRochCoulon, 2023-10-24:
Related to #1569

valentinThomazic, 2023-12-07:
Hey @flaviens, the CVA6 verification now uses a patched Verilator v5.008. I think this resolves the issue. If you agree, it would be great to close the issue !