openhwfoundation/cva6 #987: [TASK] Submit CVA6 FPGA optimizations

jquevremont, 2022-10-28:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Submit pull request(s) with CVA6 FPGA optimization to master repo:
- FPGA memory
- Front end optimization
- TLB optimization
- Cache size optimization
- Scoreboard sizing
- Regfile optimization

### Description of Done

Pull request(s) submitted
CI OK with the pull request(s)
Reviews by @jOK

### Associated PRs

First PR: #983

jquevremont, 2022-10-28:
First PR #983 submitted. More to come.

jquevremont, 2023-01-20:
Decided to break down into several small PRs (rather than a large one, tougher to review).
Optimized BTB, scoreboard configuration have been merged.
Pending PR about fifo_v3.
Upcoming PR about optimized regfile and BHT.

jquevremont, 2023-03-14:
1 PR pending.
3 PR upcoming.
Focus on frequency optimization in 2023 (mainly targetting FPGA).

jquevremont, 2023-04-18:
1 merged PR to investigate ([#1130](https://github.com/openhwgroup/cva6/pull/1130)): timing analysis.

JeanRochCoulon, 2024-05-22:
@jquevremont    What is the status of this task ?

jquevremont, 2024-05-28:
Deferring future FPGA optimizations to a new task.

closed 2024-05-28 by jquevremont