openhwfoundation/cva6 #1001: [BUG] CI key problem

sjthales, 2022-11-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello @zarubaf @JeanRochCoulon @eyssartk @jq ,

I am currently having a problem with CI tests.
[image]

There seems to be a problem with the key https://download.opensuse.org/repositories/home:phiwag:edatools/xUbuntu_20.04/Release.key in the cva6/ci/setup.sh file
[image]

Have you observed this kind of problem on your side?

XiaoAn109, 2022-11-30:
Did you mean the ci/setup.sh? Me too. And I change the curl -fsSL https:\..... to curl -fsSL http:\. Then it works

zarubaf, 2022-11-30:
@XiaoAn109 thanks! @sjthales would you mind defaulting to the http protocol until we can figure out what happened exactly?

I’ve seen that the `edatools` dep registry grew stale. I will think about what to do with it, maybe a OpenHW registry would make sense?

sjthales, 2022-11-30:
On Monday, that didn't work and yesterday, that worked (in the same environment)

XiaoAn109, 2022-11-30:
Yep. Today it works.

AnikBalo, 2022-12-29:
Probably curl is not installed in your ubuntu. Install curl: sudo apt install curl
Then try ./ci/setup.sh

MikeOpenHWGroup, 2023-02-07:
Hi @sjthales, what is the status of this issue?

closed 2023-02-14 by sjthales