openhwfoundation/cva6 #1021: [BUG] Variane_testharness: invalid option -- 'v' even if DEBUG was set to 1

TNdu, 2022-12-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,

I just cloned yesterday (15/12/22) the repo (i was working from an older but working version and wanted to started fresh) and built the verilator model with: _make verilate DEBUG=1_.

But when i try to generate some _*.vcd_ file, i get an error telling me _-v_ is not a valid option.

The command i am running is:
_XXX:~/Thesis/hardware/cva6$ ./work-ver/Variane_testharness -v test.vcd ~/Thesis/software/benchmark_pipe_stage/bin/main.elf_

I have Verilator 4.110 2021-02-25 rev UNKNOWN.REV.

It used to work and something has been changed since.

I already checked the file: _ariane_tb.cpp_ and it seems that the _VM_TRACE_ is not set to 1. But if i force it to 1 with the command : _make verilate DEBUG=1 VM_TRACE=1_ then i have this error during the compilation:

../corev_apu/tb/ariane_tb.cpp:323:10: error: ‘class Variane_testharness’ has no member named ‘trace’

Could you please help me to fix this as i need to generate the traces in vcd to continue working on the CV6A64.

Thank you

XiaoAn109, 2022-12-19:
Use make verilate TRACE_FAST=1 ，then it works ，and no need to add DEBUG=1 (You can add it).

TNdu, 2022-12-19:
Thank you for your answer. I did try to run _make verilate TRACE_FAST=1_ but i've got an other error:

```
[Verilator] Building Model
verilator -f core/Flist.cva6 /home/XXXX/Thesis/hardware/cva6/corev_apu/register_interface/src/reg_intf.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/tb/rvfi_pkg.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/tb/ariane_soc_pkg.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dm_pkg.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/tb/ariane_axi_soc_pkg.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/tb/axi_adapter.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/bootrom/bootrom.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/bootrom/dromajo_bootrom.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/clint/axi_lite_interface.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/clint/clint.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb_wrap.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb_64_32.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/apb_timer/apb_timer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/apb_timer/timer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_w_buffer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_r_buffer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_slice_wrap.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_slice.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_single_slice.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_ar_buffer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_b_buffer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/fpga/src/axi_slice/src/axi_aw_buffer.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_amos.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_atomics.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_res_tbl.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_lrsc_wrap.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_amos_alu.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_lrsc.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_atomics_wrap.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/axi_mem_if/src/axi2mem.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/rv_plic/rtl/rv_plic_target.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/rv_plic/rtl/rv_plic_gateway.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/rv_plic/rtl/plic_regmap.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/rv_plic/rtl/plic_top.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dmi_cdc.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dmi_jtag.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dmi_jtag_tap.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dm_csrs.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dm_mem.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dm_sba.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/src/dm_top.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/riscv-dbg/debug_rom/debug_rom.sv /home/XXXX/Thesis/hardware/cva6/corev_apu/register_interface/src/apb_to_reg.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_multicut.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/rstgen_bypass.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/rstgen.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/stream_mux.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/stream_demux.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/exp_backoff.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/addr_decode.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/stream_register.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_cut.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_join.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_delayer.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_to_axi_lite.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_id_prepend.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_atop_filter.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_err_slv.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_mux.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_demux.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/axi/src/axi_xbar.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/cdc_2phase.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/spill_register_flushable.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/spill_register.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/stream_arbiter.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/stream_arbiter_flushable.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/deprecated/fifo_v1.sv /home/XXXX/Thesis/hardware/cva6/vendor/pulp-platform/common_cells/src/deprecated/fifo_v2.sv [... cut ...]
make[1]: Entering directory '/home/XXXX/Thesis/hardware/cva6/work-ver'
Variane_testharness.mk:68: warning: overriding recipe for target 'verilated_vcd_c.o'
/usr/share/verilator/include/verilated.mk:241: warning: ignoring old recipe for target 'verilated_vcd_c.o'
make[1]: *** No rule to make target '/include/verilated_vcd_c.cpp', needed by 'verilated_vcd_c.o'.  Stop.
make[1]: Leaving directory '/home/XXXX/Thesis/hardware/cva6/work-ver'
make: *** [Makefile:639: verilate] Error 2
```

Thank you for your time

XiaoAn109, 2022-12-20:
Hummm, It used to works. But I think maybe there are some changes on it in the latest version...... I am working on it to figure it out... If you solve it, please tell me,.Thanks.

h0lyalg0rithm, 2022-12-27:
@XiaoAn109 I faced the same issue.You need to install verilator and set the `VERILATOR_ROOT` environment variable.Once you have it it should work.
I ended up installing verilator-4.110 as the default one didnt work.

XiaoAn109, 2022-12-27:
Thanks @h0lyalg0rithm ！It works. Let's make @TNdu informed as well.

TNdu, 2023-01-03:
> @XiaoAn109 I faced the same issue.You need to install verilator and set the `VERILATOR_ROOT` environment variable.Once you have it it should work. I ended up installing verilator-4.110 as the default one didnt work.

I installed verilator 4.110 with _apt get_. Consequently i have some of the verilator binaries located at _/usr/bin_ and the files that need to be included in _/usr/share/verilator_. Thus I would like to know how did you set your `VERILATOR_ROOT` environment variable. I tried _/usr/share/verilator_ and _/usr/bin/_ and both yeld and error because all those files should be in the same global directory to begin with.

I also tried to build verilator from sources but i can't go anywhere near the tag v1.110 from the verilator github repository. Did you manage to build verilator 4.110 from sources (and doing so, using ./configure --prefix=$VERILATOR_ROOT) ? If so could you please help me to find those sources ?

Thank you

XiaoAn109, 2023-01-03:
I built verilator from sources according to https://verilator.org/guide/latest/install.html. First, you need to install the essential prerequisites. Then clone the repo to a place eg. /opt/verilator. And try 'git pull' and 'git checkout v4.110' in the repo(may need sudo).
Specifically, we can build in this way , https://verilator.org/guide/latest/install.html#run-in-place-from-verilator-root. Add export VERTILATOR_ROOT=/opt/verilator in .bashrc, the use (sudo) autoconf and (sudo) ./configure. After that, type (sudo) make -j `nproc`. When you have done this all, the problems should be solved.

TNdu, 2023-01-03:
> I built verilator from sources according to https://verilator.org/guide/latest/install.html. First, you need to install the essential prerequisites. Then clone the repo to a place eg. [...]

Thank you for your answer. I deleted my previous message because of one idiotic typo. I was looking for the tag v1.110 instead of v4.110. I'm currently installing the v4.110 from sources with a proper VERILATOR_ROOT environment variable. 

Thank you again

TNdu, 2023-01-05:
Thank you @XiaoAn109 and @h0lyalg0rithm. Indeed when VERILATOR_ROOT environment variable and you force TRACE_FAST and DEBUG to 1 during the make verilate, it does what it used to. 

The Readme.md should be updated accordingly.

AnikBalo, 2023-01-25:
Hi, I am getting this error while I try to run “make verilate DEBUG=1 TRACE_FAST=1” command: 
  
[image]

Here is the snapshot of steps and commands that I have gone through: 

 
[image]

1.	Installed all the required packages [1-8] 
2.	Build the verilator from git [13-23] 
3.	set VERTILATOR_ROOT=/opt/verilator and RISCV=home/abalo/proj/tools/rv64 in .bashrc file. 
4.	Cloned CVA6 and run ./ci/setup.sh for required tool installation.  
5.	Then while running: [make verilate DEBUG=1 TRACE_FAST=1] for building the verilator model I was getting error.
This was done in my virtual machine [ OS: Ubuntu 20.04, RAM:8GB, Core:2]

Can anyone help? Thanks

MikeOpenHWGroup, 2023-02-07:
Hi @TNdu, thanks for your issue.  I _believe_ this issue has been resolved, so I am closing it.  Please re-open it if I misunderstood.

> The Readme.md should be updated accordingly.

Agreed.  It would be a great service to the open-source community if you created a pull-request for this.  Thank in advance!

@AnikBalo, you are asking a different question.  We try very hard to maintain a one-problem-per-issue rule.  If you are still having a problem, please open a new issue.

closed 2023-02-07 by MikeOpenHWGroup