openhwfoundation/cva6 #1031: [BUG] Issue in Verilator compile: blocked and non-blocking assignment to inp_pipe_valid in fpnew_fma.sv

rsnikhil, 2023-01-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I'm trying to compile cva6 (RV64, Sv39) under Verilator.
I encounter the following error from Verilator:

    %Error-BLKANDNBLK: /home/nikhil/Git/cva6/core/fpu/src/fpnew_fma.sv:106:59:
    Unsupported: Blocked and non-blocking assignments to same variable:
    'mkTop_HW_Side.awsteria_hw.awsteria_system.awsteria_core.core_inner_reclocked.
       cva6_core.i_cva6.ex_stage_i.fpu_gen.fpu_i.fpu_gen.i_fpnew_bulk.gen_operation_groups[0].
         i_opgroup_block.gen_parallel_slices[0].active_format.i_fmt_slice.gen_num_lanes[0].
           active_lane.lane_instance.i_fma.inp_pipe_valid_q'
    106 |   logic                  [0:NUM_INP_REGS]                 inp_pipe_valid_q;

This seems to be a valid complaint, based on the following analysis.
Looking at the cited file and line number (cva6/core/fpu/src/fpnew_fma.sv:106:59:)

There is the following blocking assignment to 'inp_pipe_valid_q[0]:

      118 |   assign inp_pipe_valid_q[0]    = in_valid_i;

Then, in the following lines:

      122 |   for (genvar i = 0; i < NUM_INP_REGS; i++) begin : gen_input_pipeline
      ...
      128 |     // Valid: enabled by ready signal, synchronous clear with the flush signal
      130 |     `FFLARNC(inp_pipe_valid_q[i+1], inp_pipe_valid_q[i], inp_pipe_ready[i],
                         flush_i, 1'b0, clk_i, rst_ni)

I found the `FFLARNC macro in two files:
  1. common/submodules/common_cells/include/common_cells/registers.svh:199
  2. core/fpu/src/common_cells/include/common_cells/registers.svh:202

Whichever one is used, the above lines (122, 130) seem to perform a non-blocking assignment to 'inp_pipe_valid_q[0]' (i.e.,
inp_pipe_valid_q[i] when i == 0 in the genvar loop.

So,
* Is this analysis correct?
* If so, what is the solution?  Simply remove the blocking assignment at
    'cva6/core/fpu/src/fpnew_fma.sv:118', since [0] has a non-blocking assignment anyway?

rsnikhil, 2023-01-13:
I found the following Verilator flag to get past this:   -Wno-BLKANDNBLK
but I'm nervous about whether this is the right solution; I hope someone can advise.

zarubaf, 2023-01-13:
I think the `verilator` flag is the right thing to do (at least I remember that in the past). The code is actually correct because only index `[i+1]` are assigned non-blocking and `[0]` is assigned blocking. The problem is that `verilator` apparently doesn't understand that those are independent signals.

rsnikhil, 2023-01-26:
Ok, thanks.

closed 2023-01-26 by rsnikhil