openhwfoundation/cva6 #1033: Running cva6 on Nexys-A7

UmerShahidengr, 2023-01-20:
Hello, I am trying to create bitstream of cva6 for [Nexys-A7 FPGA Board](https://digilent.com/shop/nexys-a7-fpga-trainer-board-recommended-for-ece-curriculum/). I have set up the whole project according to the IP Sources of Nexys-A7 and have successfully synthesized it.
On implementation, I am getting the following error: 
> [Place 30-378] Input pin of input buffer i_ddr/u_xlnx_mig_7_ddr3_mig/u_iodelay_ctrl/diff_clk_ref.u_ibufg_clk_ref has an illegal connection to a logic constant value.
> [Place 30-99] Placer failed with error: 'IO Clock Placer stopped due to earlier errors. Implementation Feasibility check failed.

Can somebody guide me on how to rectify these errors?

christian-lanius, 2023-01-23:
Out of the box, no support for the Nexys A7 board is available. Note that you have to change quite a bit to get that to work.

Did you modify the FPGA source (ariane_xilinx.sv) with a new ifdef for your board to then instantiate a modified MIG? The MIGs included with the project are board specific and if you want to use a different board, you will have to add the appropriate mig.prj (you can get that from Digilent). One of the difficulties is that, different to all supported boards, the Nexys A7 only has DDR2 and a single ended clock. Thus you get the error message you see (I presume): The DDR MIG wants to be connected to a differential clock pair for its reference clock (as thats how it is for the genesys board), but you don't have such a clock in your design, one of the pins for that differential input is fixed to a constant.

I have the CVA6 running on that board, but I have not shared the required changes so far. It works, but the other thing to note is that the "big" Config of the design does not fit on the board, you will have to use one of the smaller ones, such as the one optimized for FPGAs.

UmerShahidengr, 2023-01-24:
Yes. I have modified the FPGA source file (ariane_xilinx.sv) and created a new .svh file containing all helping macros for NexysA7 board. I have modified MIG configuration according to DDR2 and added that in my project.

UmerShahidengr, 2023-01-24:
I have made the [following changes in the repo](https://github.com/openhwgroup/cva6/compare/master...UmerShahidengr:cva6:master) to run cva6 on NexysA7. 
Is there anything else required to be done? I am still stuck at the error above.

UmerShahidengr, 2023-01-24:
@christian-lanius , have you run this core on NexysA7. If yes, then which ariane version have you used for synthesis? I have resolved the input buffer error by setting MIG clk in single-ended mode. But now I am getting the following error:
> ERROR: [Place 30-640] Place Check : This design requires more Slice LUTs cells than are available in the target device. This design requires 66121 of such cell types but only 63400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. 
> ERROR: [Place 30-640] Place Check : This design requires more LUT as Logic cells than are available in the target device. This design requires 64746 of such cell types but only 63400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. 

Its saying that Slice LUTs of NexysA7 are not enough to implement the latest version of cva6.
Thanks for your help.

christian-lanius, 2023-01-26:
I have used version 6deffb27d7f031341e33e84c422a19e39095aa6c and ran that successfully on the FPGA. You have to set the TARGET_CFG variable during the make command (i.e. make fpga BOARD=YOURBOARDNAME TARGET_CFG=cv32a6_ima_sv32_fpga). You do not have to use the specific fpga one, but that one should be small enough to fit on the Nexys Board. You can also try to further modify the config_pkg to reduce utilization (in core/include). Depending on your TARGET_CFG (which defaults to cv64a6_imafdc_sv39), one of those files contains some definitions. As you arae only barely not fitting on the FPGA, choosing the FPGA config should be enough. Otherwise remove parts which you don't need (for me that would be the MMU for example).

MikeOpenHWGroup, 2023-02-07:
@christian-lanius, @UmerShahidengr.   AFAIK, it will not be possible to support the CVA6 on the Nexys A7 as the FPGA on that board is probably not big enough and the Nexys board probably does not support the required hardware.   In any case, we do not support this platform.

closed 2023-02-07 by MikeOpenHWGroup