openhwfoundation/cva6 #1035: Reproducing Dhrystone performance numbers

jwright6323, 2023-01-25:
Hi all, this might be a duplicate of a question already asked (#580), but we are trying to correlate DMIPS/MHz numbers published in this paper: https://arxiv.org/pdf/1904.05442.pdf . We are getting a lower number when we run with the cache and branch predictor settings listed in the paper, so we're assuming there's some compiler discrepancy. Is there documentation on how the DMIPS/MHz score of 1.65 was generated?

Thanks
John

MikeOpenHWGroup, 2023-02-07:
Hi @zarubaf, I honestly do not know who can answer this question, so you get it.  :wink:

cathales, 2023-10-20:
Hello,

FYI here is how we run Dhrystone in regressions https://github.com/openhwgroup/cva6/blob/master/verif/regress/dhrystone.sh

With this script we get 1.3 DMIPS/MHz (click the "PASS" button on `master` pipelines to get details): https://riscv-ci.pages.thales-invia.fr/dashboard/

This might not be the best configuration for performance as this test is only run to detect performance changes in PRs.

closed 2025-01-06 by jwright6323