openhwfoundation/cva6 #1037: Support of the H-extension - Roadmap ?

pcotret, 2023-01-26:
https://cva6.readthedocs.io/en/latest/specifications/cva6_requirement_specification.html?highlight=PVL-60#privileges-and-virtual-memory

Requirement PVL-60 deals with the H-extension:
>  PVL‑60 | CV64A6 shall support as an option the H extension (hypervisor) version 1.0.

However, it doesn't seem to be implemented in the CVA6 yet. Is there a roadmap for the H-extension support in the CVA6 ?

jquevremont, 2023-02-08:
This is a feature developed by the university of Minho in a separate branch. They need some work to make it optional before we can integrate it into the master branch. Pinging @sandro2pinto to get an update.

sandro2pinto, 2023-02-08:
Hello all,

Since the OpenHW Group Face-to-Face @ RISC-V Summit 22 and the presentation of the CVA6-H (CVA6 with Hypervisor Extension support) Project Concept Proposal, we were waiting for:

1. The stabilization of the methodology being conceived internally at the OpenHW Group on how to include optional features in maintained cores;

2. Feedback on the Project Concept and interest to advance for the Project Launch.

We are ready to upstream, but we need concrete instructions for #1 (and Bruno has been attending the CVA6 meetings).

jquevremont, 2023-04-19:
Follow up: PR #1112 is currently open.

pcotret, 2023-04-19:
I think it's time to close this issue. Thanks for the news @jquevremont :)

closed 2023-04-19 by pcotret