openhwfoundation/cva6 #1048: Incorrect documentation of `misa` CSR in User Manual

MikeOpenHWGroup, 2023-02-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

There are several issues related to `misa` documentation:
1. The "access type" should be RO, not RW.
2. For CV32A6* the "reset value" should be: {2'b01, 4'b0000, 26'b01_0100_1000_0001_0001_0000_0100} or 0x4148_1104.  That is MXL=32-bits, Non-standard instructions, User mode, Supervisor mode and MIC ISA.

Not: I could be wrong about the power-on-reset value of misa, but for sure it is not 0x0.

spidugu444, 2023-02-22:
HI @MikeOpenHWGroup 

While verifying Access mode for MISA CSR I found that default value for that CSR is  0x4014_1104.
where **Non-standard extensions present** has not been enabled.
could you please confirm this from your end.

MikeOpenHWGroup, 2023-03-06:
Hello @jquevremont.  You are the "keeper" of the CVA Requirements, so I think this question is you for to decide.  I assumed that since CVA6 supports a non-standard [fence.t](https://docs.openhwgroup.org/projects/cva6-user-manual/02_cva6_requirements/cva6_requirements_specification.html#fence-t-custom-instruction) instruction that it should set the `Non-standard extensions present` bit in the MISA CSR.

What do you think?

jquevremont, 2023-03-07:
`fence.t` is not yet implemented in OpenHW repo (but in @niwis's)  and we have not yet identified contributors. So I suggest to state that we have no non-standard extensions, for consistency with the rest of the design. The bit will need to be changed when we have `fence.t`.

commit 6ce4705a (2023-08-08): Fix the reset value of MISA in user manual (fix #1048) (#1330)
closed 2023-08-08 by jquevremont