openhwfoundation/cva6 #1053: CVA6-specific behaviour for WLRL CSR fields

MikeOpenHWGroup, 2023-02-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Section 2.3 of the RISC-V Privileged Specification (20190608) says:
```
Implementations are permitted but not required to raise an illegal instruction exception if an
instruction attempts to write a non-supported value to a WLRL field. Implementations can
return arbitrary bit patterns on the read of a WLRL field when the last write was of an illegal
value, but the value returned should deterministically depend on the illegal written value and the
value of the field prior to the write.
```
It is my understanding that: the CVA6 does _not_ raise an illegal instruction exception if an instruction attempts to write a non-supported value to a WLRL field.

Please confirm if this is true or not.   Pending the closure of this issue, I will create an update to the User Manual.

zarubaf, 2023-02-09:
That is correct

JeanRochCoulon, 2023-03-14:
@yanicasa will update the CSR specification in that way.

JeanRochCoulon, 2025-01-02:
@zchamski We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case?

zchamski, 2025-01-09:
The `riscv-config` spec of CV32A65X explicitly specifies `unchanged` as the **complete** behavior for illegal value writes of all CSRs and CSR fields.

Spike and the RTL behave in the same way as well, so the manual can be updated accordingly.

zchamski, 2025-01-09:
@MikeOpenHWGroup: I think you can close this issue.

JeanRochCoulon, 2025-01-09:
Before closing, [the ](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html#_writeread_only_legal_values_wlrl) must be updated.

zchamski, 2025-01-23:
@ASintzoff, @JeanRochCoulon : what is the status of the documentation update?

JeanRochCoulon, 2025-01-23:
As this issue requires a specification update, I assign @ASintzoff

commit cbb08e8d (2025-01-23): docs: clarify WLRL CSR fields on CVA6 (fix #1053) (#2733)

as priv-csrs.adoc was not yet tailored for CVA6, the file is copied
and tailored
closed 2025-01-23 by JeanRochCoulon