openhwfoundation/cva6 #1054: [BUG] Variane_testharness: invalid option -- 'v' even if DEBUG was set to 1

ewoo5, 2023-02-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I just cloned the repo today (02/08/2023), and I have verilator version 4.110 2021-02-25 rev v4.110.
After running `make verilate DEBUG=1`, running `work-ver/Variane_testharness -v hello.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf` does not work. 
The -v option does not work as this is the output: `work-ver/Variane_testharness: invalid option -- 'v'`.
If I try appending the options at the end instead: 
`work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf -v hello.vcd`
It runs but does not produce any .vcd files. So, I guess the options are just ignored.

I realized that this issue has been mentioned before in issue  #1021 .
So, I tried `make verilate TRACE_FAST=1 DEBUG=1`, and I already have set  $VERILATOR_ROOT variable.
And like @AnikBalo, the make fails with these errors at the end:
`
riane_testharness__Trace__6__Slow.o Variane_testharness__Trace__6__Slow.cpp
ccache g++  -I.  -MMD -I/home/ewoo5/ece498sjp/verilator/include -I/home/ewoo5/ece498sjp/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/include -I/home/ewoo5/ece498sjp/cva6/riscv-toolchain/include -I/home/ewoo5/ece498sjp/cva6/riscv-toolchain/include  -std=c++11 -I../corev_apu/tb/dpi -O3  -DVL_DEBUG  -DVL_THREADED -std=gnu++14  -c -o Variane_testharness__Trace__7__Slow.o Variane_testharness__Trace__7__Slow.cpp
ccache g++  -I.  -MMD -I/home/ewoo5/ece498sjp/verilator/include -I/home/ewoo5/ece498sjp/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/include -I/home/ewoo5/ece498sjp/cva6/riscv-toolchain/include -I/home/ewoo5/ece498sjp/cva6/riscv-toolchain/include  -std=c++11 -I../corev_apu/tb/dpi -O3  -DVL_DEBUG  -DVL_THREADED -std=gnu++14  -c -o Variane_testharness__Trace__8__Slow.o Variane_testharness__Trace__8__Slow.cpp
ccache g++  -I.  -MMD -I/home/ewoo5/ece498sjp/verilator/include -I/home/ewoo5/ece498sjp/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/include -I/home/ewoo5/ece498sjp/cva6/riscv-toolchain/include -I/home/ewoo5/ece498sjp/cva6/riscv-toolchain/include  -std=c++11 -I../corev_apu/tb/dpi -O3  -DVL_DEBUG  -DVL_THREADED -std=gnu++14  -c -o Variane_testharness__Trace__9__Slow.o Variane_testharness__Trace__9__Slow.cpp
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__12__Slow.o] Error 1
make[1]: *** Waiting for unfinished jobs....
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__Trace__2__Slow.o] Error 1
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__11__Slow.o] Error 1
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__Trace__Slow.o] Error 1
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__20__Slow.o] Error 1
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__22__Slow.o] Error 1
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[1]: *** [/home/ewoo5/ece498sjp/verilator/include/verilated.mk:238: Variane_testharness__6__Slow.o] Error 1
make[1]: Leaving directory '/home/ewoo5/ece498sjp/cva6/work-ver'
make: *** [Makefile:639: verilate] Error 2
`

Any idea what is wrong? I cannot do much without being able to dump .vcd files.

Thanks

ASintzoff, 2023-02-24:
With CVA6 at commit 44817bd (latest available today on master branch), I'm able to build `work-ver/Variane_testharness` with `make verilate TRACE_FAST=1 DEBUG=1` using `Verilator 4.110 2021-02-25`.

Unfortunately, as I'm not able to reproduce your issue, it will be difficult to provide you answers.

Nevertheless, once the build stops, do you have any files in the `work-ver` directory? If yes, what is the more recent one?

ewoo5, 2023-02-26:
There is a lot of flies in the work-ver directory, and this is just a snippet of some of the newest files after make fails:

[image]

Also, I am not sure if this matters, but I am using WSL2.

ChuhengZhou, 2023-04-11:
@ewoo5 @hhhsiang, I think this problem is caused by exhausting the cores or memory when using wsl2. After setting the wsl2 memory and swap size to 12GB and 16GB respectively, and deleting the `-j${NUM_JOBS}` at line 640 of `<cva6>/Makefile` , the error disppears.

@hhhsiang To reinstall verilator, I change the verilator version in `<cva6>/ci/install-verilator.sh` to 4.110,  then set `$VERILATOR_ROOT` to the location you want verilator to be installed, finally run the script.

JeanRochCoulon, 2024-02-14:
Hi guys,
The simulation commands and the verilator version you use is not aligned on the recommandations done in README of the project. If you are aligned on the README, support cannot be given. I will close the issue.

closed 2024-02-14 by JeanRochCoulon