openhwfoundation/cva6 #1058: [BUG] <Wrong assignment of X_NUM_RS>

AyoubJalali, 2023-02-13:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
I notice that in the cvxif.pkg, the **X_NUM_RS** is assign by  **NR_RGPR_PORTS** (a constant in the ariane_pkg.sv), hence the **X_NUM_RS**, can't change only if it changed in the ariane_pg.sv.
So what do you think about this ?

Gchauvon, 2023-02-13:
Hello, 
I don't think it is a bug. It is a choice I made when implementing CVXIF in order to not increase size of the register file when cvxif does not need NR_RGPR_PORTS=3. Like a lot of parameters in ariane_pkg.sv, you can modify it. It takes 2 or 3 for value. and both parameters together to avoid miss combination. 
Maybe it is not the best of implementation, feel free to recommend a better one if you think it is necessary.

AyoubJalali, 2023-02-13:
The problem is that if I wanna execute a test with 3 source registers, i should change in the **araine_pkg**, it's like i change in the RTL, and i don't think it's practical.

Gchauvon, 2023-02-13:
Yes if you want to use 3 source registers, you have to change the RTL because it is not the default configuration of CVA6.

MikeOpenHWGroup, 2023-02-17:
Hi @Gchauvon, what is the proper value of `NR_RGPR_PORTS` for the CV32A6 v5.0.0?

Gchauvon, 2023-02-17:
It's `NR_RGPR_PORTS=2`

Gchauvon, 2023-12-06:
We could add this parameters in the config_pkg. It can be part of CVA6 parametrization.

AyoubJalali, 2024-12-24:
The protocol has been changed, so this issue is no more valid

closed 2024-12-24 by AyoubJalali