openhwfoundation/cva6 #1060: Incorrect documentation of mtvec CSR in user manual

spidugu444, 2023-02-14:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

This issue points out **mtvec** documentation. 
In CVA6 CSR documentation, Mode field access for **mtvec** is mentioned as RW as shown below.
<html>
<body>

Bits | Name | Display Name | Access Type | Reset
-- | -- | -- | -- | --
[31:2] | BASE |   | RW | 0b0
[1:0] | MODE |   | RW | 0b0

</body>
</html>
The access type for the Mode field should be as mentioned below. 

<html>
<body>

Bits | Mode | Description
-- | -- | -- |
1 | RO | MODE[1]: always 0
0 | RW | MODE[0]: 0 = direct mode, 1 = vectored mode.

</body>
</html>

JeanRochCoulon, 2023-02-14:
Thank you @spidugu444 for this GitHub issue.
As it is connected with the previous discussion involving @ASintzoff and @MikeOpenHWGroup about https://github.com/openhwgroup/cva6/pull/1049, I prefer to assign it to André

MikeOpenHWGroup, 2023-02-16:
Hi @JeanRochCoulon.  It is good that you have an opinion about who to assign this to.  In such cases, please feel free to change the Assignee.

Hi @spidugu444.  Please ensure that Component and Status flags are added to each Issue.

commit b928fdfa (2023-07-31): Adding Legal values to MTVEC and STVEC registers to fix #1060 and #1079 issues (#1325)

* Adding Legal values to MTVEC and STVEC registers to fix #1060 and #1079 issues

* Updating MTVEC and STVEC description

---------
closed 2023-07-31 by ASintzoff