openhwfoundation/cva6 #1063: cebreak test failure in CVA6 riscv-arch-test suite

spidugu444, 2023-02-14:
Hi All,

**In CVA6 riscv-arch-test suite cebreak test is failing in both 32 and 64 bits due to gpr value mismatch between spike and verilator.**
can we know why this mismatch is happening between spike and verilator.

please find the steps to reproduce issue.
1.git clone https://github.com/openhwgroup/core-v-verif.git
  git checkout cv32a6_v0.1.0/deprecated
2. SET RISCV varible to toolchain.
3. Enable cebreak iteration to 1 in **cva6/tests/testlist_riscv-arch-test-cv32a60x.yaml** for (32-bits)
                                                       **cva6/tests/testlist_riscv-arch-test-cv64a6_imafdc_sv39.yaml** (64-bits)
4.command to run 64-bit tests **source cva6/regress/dv-riscv-arch-test.sh**
5. For running 32-bit test we need to set DV_TARGET to cv32a60x as shown below
    **export DV_TARGET=cv32a60x**
    command to run 32-bit riscv-arch-test  **source cva6/regress/dv-riscv-arch-test.sh**

Here are the corresponding csv and log files for both 32 and 64 bits as attached below.

cebreak 64-bit csv and log files
[veri_cebreak-01_64_bit (2).log](https://github.com/openhwgroup/cva6/files/10731940/veri_cebreak-01_64_bit.2.log)
[veri_cebreak-01_64_bit (2).csv](https://github.com/openhwgroup/cva6/files/10731942/veri_cebreak-01_64_bit.2.csv)
[spike_cebreak-01_64_bit (1).log](https://github.com/openhwgroup/cva6/files/10731943/spike_cebreak-01_64_bit.1.log)
[spike_cebreak-01_64_bit (1).csv](https://github.com/openhwgroup/cva6/files/10731945/spike_cebreak-01_64_bit.1.csv)

cebreak 32-bit csv and log files

[32-bit_spike_cebreak-01 (2).log](https://github.com/openhwgroup/cva6/files/10731954/32-bit_spike_cebreak-01.2.log)
[32-bit_veri_cebreak-01 (2).log](https://github.com/openhwgroup/cva6/files/10731955/32-bit_veri_cebreak-01.2.log)
[32-bit_spike_cebreak-01 (2).csv](https://github.com/openhwgroup/cva6/files/10731957/32-bit_spike_cebreak-01.2.csv)
[32-bit_spike_cebreak-01 (1).log](https://github.com/openhwgroup/cva6/files/10731958/32-bit_spike_cebreak-01.1.log)

JeanRochCoulon, 2023-02-14:
Hello @spidugu444,
It is great to give us the way to reproduce the failing case. But can you explain the problem?   According to your first investigation, we can determine the best person to re-assign the problem.

MikeOpenHWGroup, 2023-02-16:
Also, please ensure that `Component`, `Type` and `Status` flags are added to each Issue.  Thanks!

ASintzoff, 2023-11-13:
`mtval` value is not the same on Spike and on CVA6. On Spike, it is the `c.ebreak` instruction address (`0x800003ac` on CVA64 or `0x80000114` on CVA32) and on CVA6, it is the `c.ebreak` instruction itself (`0x9002`).

It has also been reported in https://github.com/openhwgroup/cva6/issues/898

closed 2023-11-13 by ASintzoff