openhwfoundation/cva6 #1070: boot linux on verilator

rapiz1, 2023-02-16:
Is it possible to boot linux on ariane_testharness and verilator? How to make the image that can be directly loaded into ram? And what changes should be done to the cva6 source?

MikeOpenHWGroup, 2023-02-16:
Hi @rapiz1, the short answer to your question is "yes".  I am not sure if we have documented the flow for that.  @sjthales can you comment?

jyrj, 2023-03-14:
Is there any update for this? I like to see how we could use the cva6-sdk with the verilator model.

retina007, 2023-05-25:
https://github.com/openhwgroup/programs/blob/master/TGs/cores-task-group/cva6/cva6_9_4_21.md
say:
Sebastien has been successful booting Linux-64 CV64A6 using Questa.
@MikeOpenHWGroup 
@sjthales

retina007, 2023-05-25:
@MikeOpenHWGroup 
@JeanRochCoulon also noted that they are booting Linux way faster on Verilator .  can merge booting Linux-64 CV64A6 using Verilator to cva6 master. thanks

JeanRochCoulon, 2023-05-25:
For the time we are able to boot Linux on FPGA in Thales CI, we stopped the execution of inux with Verilator, maybe 18 months ago. But you cn find some information to how to boot Linux in ./cva6/regress/linux.sh from core-v-verif repository.

retina007, 2023-06-01:
@JeanRochCoulon Hi.
1. I run ./cva6/regress/linux.sh(veri-testharness) from core-v-verif repository. 
2. I use the bbl(https://github.com/openhwgroup/cva6-sdk/releases/tag/v0.3.0)
3. maybe not important below, because use preload elf.
(I use riscv-toolchain **corev-openhw-gcc-centos7-20211104/bin/riscv32-corev-elf-nm** ,yes, 64 and 32 is the same.
so the makefile (https://github.com/openhwgroup/core-v-verif/blob/master/cva6/sim/Makefile) change:
veri-testharness:
	make -C $(path_var) verilate verilator="verilator --no-timing" target=$(target) defines=$(subst +define+,,$(isscomp_opts))
	$(path_var)/work-ver/Variane_testharness $(if $(TRACE_COMPACT), -f verilator.fst) $(if $(TRACE_FAST), -v verilator.vcd) $(elf) $(issrun_opts) +PRELOAD=$(elf) \
	  +tohost_addr=$(shell $$RISCV/bin/**riscv32-corev-elf-nm** 
)

4. but terminal stop at time out 40000000(defined at linux.sh, about 4 hours simulation), nothing print out on booting linux(exclude last line log "Listening on port 42831").
5. other cva6/regress test success,  can help to pass linux.sh
thanks

JeanRochCoulon, 2023-06-01:
40000000 corresponds to the CVA6 RTL timeout. To go further, you have to inclement it. Tell me if better !

retina007, 2023-06-05:
@JeanRochCoulon 
Hi!  I run 3 days about  73776208 lines/5GB (trace_hart_00.dasm  show below)
but case linux.sh seem cannot stop. terminal also print nothing. 
(only print:  +tohost_addr=0000000080008008
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 42763)
Can know what is going on from the command below?
trace_hart_00.dasm:
                 265 0x10000 M (0x00100413) DASM(00100413)
                 270 0x10004 M (0x01f41413) DASM(01f41413)
                 275 0x10008 M (0xf1402573) DASM(f1402573)
                 280 0x1000c M (0x00000597) DASM(00000597)
                 285 0x10010 M (0x07458593) DASM(07458593)
                 290 0x10014 M (0x00040067) DASM(00040067)
                 300 0x80000000 M (0x1f80006f) DASM(1f80006f)
                 306 0x800001f8 M (0x00000093) DASM(00000093)
                 308 0x800001fc M (0x00000113) DASM(00000113)
                 313 0x80000200 M (0x00000193) DASM(00000193)
                 315 0x80000204 M (0x00000213) DASM(00000213)
                 316 0x80000208 M (0x00000293) DASM(00000293)
                 317 0x8000020c M (0x00000313) DASM(00000313)
                 322 0x80000210 M (0x00000393) DASM(00000393)
                 324 0x80000214 M (0x00000413) DASM(00000413)
                 325 0x80000218 M (0x00000493) DASM(00000493)
                 326 0x8000021c M (0x00000613) DASM(00000613)
                 331 0x80000220 M (0x00000693) DASM(00000693)
                 333 0x80000224 M (0x00000713) DASM(00000713)
                 334 0x80000228 M (0x00000793) DASM(00000793)
                 335 0x8000022c M (0x00000813) DASM(00000813)
                 340 0x80000230 M (0x00000893) DASM(00000893)
                 342 0x80000234 M (0x00000913) DASM(00000913)
                                                              1,18          Top
……

              427440 0x800021d6 M (0x00008f55) DASM(00008f55)
              427441 0x800021d8 M (0x30071073) DASM(30071073)
              427448 0x800021dc M (0x0000777d) DASM(0000777d)
              427449 0x800021de M (0x0000978a) DASM(0000978a)
              427453 0x800021e0 M (0x00008ff9) DASM(00008ff9)
              427455 0x800021e2 M (0xec078793) DASM(ec078793)
              427456 0x800021e6 M (0x34079073) DASM(34079073)
              427458 0x800021ea M (0x34151073) DASM(34151073)
              427460 0x800021ee M (0x0000852e) DASM(0000852e)
              427462 0x800021f0 M (0x000085b2) DASM(000085b2)
              427464 0x800021f2 M (0x30200073) DASM(30200073)
              427477 0x80200000 S (0x10401073) DASM(10401073)
              427479 0x80200004 S (0x00ac2197) DASM(00ac2197)
              427481 0x80200008 S (0xb1418193) DASM(b1418193)
              427482 0x8020000c S (0x00006299) DASM(00006299)
              427486 0x8020000e S (0x1002b073) DASM(1002b073)
              427494 0x80200012 S (0xf8818693) DASM(f8818693)
              427495 0x80200016 S (0x00004605) DASM(00004605)
              427505 0x80200018 S (0x00c6a6af) DASM(00c6a6af)
              427515 0x8020001c S (0x0000e2e9) DASM(0000e2e9)
              427516 0x8020001e S (0x00ac4697) DASM(00ac4697)
              427518 0x80200022 S (0xfe268693) DASM(fe268693)
              427519 0x80200026 S (0x00b8b717) DASM(00b8b717)
                                                              251455,32      0%
......

             1426817 0x802000a0 S (0xf6460613) DASM(f6460613)
             1426819 0x802000a4 S (0x00008231) DASM(00008231)
             1426820 0x802000a6 S (0xfff0059b) DASM(fff0059b)
             1426822 0x802000aa S (0x03f59593) DASM(03f59593)
             1426823 0x802000ae S (0x00008e4d) DASM(00008e4d)
             1426826 0x802000b0 S (0x0001c517) DASM(0001c517)
             1426828 0x802000b4 S (0xf5050513) DASM(f5050513)
             1426830 0x802000b8 S (0x00008131) DASM(00008131)
             1426832 0x802000ba S (0x00008d4d) DASM(00008d4d)
             1426833 0x802000bc S (0x12000073) DASM(12000073)
[... 21 lines cut ...]
            34360338 0xffffffe0007daac2 S (0x0000fc06) DASM(0000fc06)
            34360339 0xffffffe0007daac4 S (0x0000ec4e) DASM(0000ec4e)
            34360340 0xffffffe0007daac6 S (0x0000e852) DASM(0000e852)
            34360340 0xffffffe0007daac8 S (0x00000080) DASM(00000080)
            34360341 0xffffffe0007daaca S (0x000084aa) DASM(000084aa)
            34360342 0xffffffe0007daacc S (0x0000892e) DASM(0000892e)
            34360364 0x80000004 M (0x34011173) DASM(34011173)
            34360367 0x80000008 M (0x1a010863) DASM(1a010863)
            34360375 0x8000000c M (0x04a13823) DASM(04a13823)
            34360380 0x80000010 M (0x04b13c23) DASM(04b13c23)
            34360381 0x80000014 M (0x342025f3) DASM(342025f3)
            34360384 0x80000018 M (0x0805d263) DASM(0805d263)
            34360390 0x8000009c M (0x00113423) DASM(00113423)
            34360396 0x800000a0 M (0x00313c23) DASM(00313c23)
            34360398 0x800000a4 M (0x02413023) DASM(02413023)
            34360399 0x800000a8 M (0x02513423) DASM(02513423)
            34360399 0x800000ac M (0x00009297) DASM(00009297)
            34360405 0x800000b0 M (0x02613823) DASM(02613823)
                                                              15719676,32   21%

......
          147414715 0xffffffe00061552a S (0x000066b1) DASM(000066b1)
           147414717 0xffffffe00061552c S (0xa7e68693) DASM(a7e68693)
           147414718 0xffffffe00
                                                              73776208,12   Bot

JeanRochCoulon, 2023-06-06:
Yes, of course the Linux execution does not stop. Executing the whole boot means executing 400M cycles. Too much for Verilator. That's the reason why we use FPGA to verify Linux boot. But thank to your trace, you can see/analyze the Linux start boot execution.

retina007, 2023-06-08:
@JeanRochCoulon 
4.Another question, if the simulation of linux.sh does not stop, what signs can prove that the simulation of booting Linux is almost successful?

JeanRochCoulon, 2025-01-02:
@retina007 We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

retina007, 2025-01-02:
@JeanRochCoulon Close the Issues. Thanks

closed 2025-01-02 by JeanRochCoulon