openhwfoundation/cva6 #1073: execute work-ver/Variane_testharness rv64um-v-divuw

hhhsiang, 2023-02-17:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I have error after executing the command (work-ver/Variane_testharness rv64um-v-divuw)：
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

I also tried this：https://github.com/openhwgroup/cva6/issues/902
I executed the command：make run-asm-tests-verilator
,then it showed：

make[1]: Leaving directory '/home/aislab42190/cva6/work-ver'
work-ver/Variane_testharness tmp/riscv-tests/build/isa//rv64ui-p-add
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 44235
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.
make: *** [Makefile:645: rv64ui-p-add-verilator] Error 255

I tried other ways from existing issues with similar bug, but I still could not fix it.
///////////////////////////////////////////////////////////////////////////////////////////////////////////
Besides, I found that there is no "share" folder under $RISCV/riscv64-unknown-elf/
As a result, when I use this command as mentioned above：
./Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
it just said "No such file or directory."
How can I fix it?
Thanks for helping!

hhhsiang, 2023-02-21:
The followings are the steps I build the environment：
At first, I used these commands to checkout the repository under /home/aislab42190/
1. git clone https://github.com/openhwgroup/cva6.git
2. cd cva6
3. git submodule update --init --recursive

after that, I build a new directory named "RISCV" under /home/aislab42190/
then I set the environment variable by these commands：
6. export RISCV="/home/aislab42190/RISCV/"
7. export PATH=$PATH:$RISCV/bin
8. source ~/.bashrc

to install all required tools：
9. ./ci/setup.sh
10. make verilate
11. make verilate DEBUG=1

after the above commands were executed, I use the command：
work-ver/Variane_testharness rv64um-v-divuw

then it showed：
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36327
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

Thank you very much！

mackncheesiest, 2023-02-22:
Pretty sure that second argument needs to be a RISCV ELF binary, and no binary with that name exists in the top level repository folder. It looks like it exists under `./tmp/riscv-tests/build/isa/rv64um-v-divuw`, so running

`./work-ver/Variane_testharness ./tmp/riscv-tests/build/isa/rv64um-v-divuw`

Seems to have worked for me.

hhhsiang, 2023-02-23:
Hi, @mackncheesiest 
It's all good now! Besides, May I ask what is "-v-divuw" mean after "rv64um"?
Thanks for your help!

mackncheesiest, 2023-02-23:
Not sure! I was just looking at this repository to help another student in our lab who ran into the same issue :)

ASintzoff, 2023-02-24:
> Besides, may I ask what is "-v-divuw" mean after "rv64um"?

rv64: 64-bit
u: user mode
m: M extension
v: virtual
divuw: divuw instruction

hhhsiang, 2023-02-24:
Hi @mhafiz95 , I don't seem to have encountered this issue before. What command did you use when you encountered this problem?

hhhsiang, 2023-02-24:
@ASintzoff Thank you very much!

Do we need to check the 'trace_hart_00.dasm' file to determine the operating status? Or can we use other tools to help us view the waveform so that we don't have to read the file line by line? According to the top-level [README](https://github.com/openhwgroup/cva6#install-verilator-simulation-flow), we appear to be able to use the 'spike' tool by running the command `spike-dasm < trace_hart_00.dasm > logfile.txt`, but when I tried it, I received an error message saying 'command not found'.

hhhsiang, 2023-03-01:
@mhafiz95 
I'm facing the same error and trying to find a solution for it.
Please let me know If you find the way to solve it.

ASintzoff, 2023-03-02:
> According to the top-level [README](https://github.com/openhwgroup/cva6#install-verilator-simulation-flow), we appear to be able to use the 'spike' tool by running the command `spike-dasm < [...]

`spike-dasm` is one of the executable files provided with `spike`. To get it, you need to install `spike` [https://github.com/riscv-software-src/riscv-isa-sim](https://github.com/riscv-software-src/riscv-isa-sim).

hhhsiang, 2023-03-05:
hi @ASintzoff , I executed the `install-spike.sh` command found in the 'ci' folder, but I am still unable to use `spike-dasm`. I am wondering if there is a different installation process than the one you provided, or if I executed the command in the wrong path. Besides, the 'trace_hart_00.dasm' file is located in the 'cva6' folder, so I also executed the command in that folder. Could you please confirm if this is the correct path to execute the command? Thank you for your help!

ASintzoff, 2023-03-06:
`spike-dasm` executable should be located in the same directory as `spike` one and according to `ci/install-spike.sh`, it should be under `$RISCV` directory. So the location depends on the value of `$RISCV`.

ASintzoff, 2023-10-31:
As no update since several months, issue closed.

Feel free to reopen it if new information.

closed 2023-10-31 by ASintzoff