openhwfoundation/cva6 #1079: Incorrect documentation of MIE, MIP, SIP, SIE and STVEC CSR'S in CV32A6_CSR manual

spidugu444, 2023-02-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

**STVEC:**
This issue points out the STVEC documentation.
In CV32A6_CSR documentation, Mode field access for mtvec is mentioned as RW as shown below.

Bits | Name | Display Name | Access Type | Reset
-- | -- | -- | -- | --
[31:2] | BASE |   | RW | 0b0
[1:0] | MODE |   | RW | 0b0

</body>
</html>
As per riscv-privilleged specification Mode field should be as mention below.

For stvec CSR mode field we have two implementation options i.e Direct Mode(00) , vector Mode (01) and for remaining implementations i.e for 10,11 has been reserved for future use.
so the mode[1] bit is fixed to 0 making this bit to RO.

<html>
<body>

Bits | Mode | Description
-- | -- | --
1 | RO | MODE[1]: always 0
0 | RW | MODE[0]: 0 = direct mode, 1 = vectored mode.

</body>
</html>

**MIE:**
As per riscv-privileged specification for Machine Interrupt Enable(MIE) CSR  bits 0,2,4,6,8,10,12-31 has been fixed to 0 and access type should be RO.
In CV32A6_CSR document for MIE  CSR bits 0,4,8 has been marked as RW which will be RO.

**MIP:**
As per riscv-privileged specification for Machine Interrupt Pending(MIP) CSR  bits 0,2,4,6,8,10  has been fixed to 0 and access type should be RO.
In CV32A6_CSR document for MIP CSR bits 0,4,8 has been marked as RW which will be RO.

**SIP:**
As per riscv-privileged specification for Supervisor Interrupt Pending(SIP) CSR  bits 0,2-4,6-8  has been fixed to 0 and access type should be RO.
In CV32A6_CSR document for SIP CSR bits 0,4,8 has been marked as RW which will be RO.

**SIE:**
As per riscv-privileged specification for Supervisor Interrupt Enable(SIE) CSR  bits 0,2,4,6,8,10  has been fixed to 0 and access type should be RO.
In CV32A6_CSR document for SIE CSR bits 0,4,8 has been marked as RW which will be RO.

JeanRochCoulon, 2023-03-14:
Hello @yanicasa 
Can you check whether this point is well specified in riscv-arc-test repository. If the scope is to generate the CSR specification from this riscv-arch-test description, the issue will be solved.

spidugu444, 2023-04-26:
As per this [PR](https://github.com/openhwgroup/cva6/pull/1074) default value for **MVENDORID** CSR should be 0x0602, which needs to be update on Design Document.

frikhaAziz, 2023-08-03:
I added legal values to STVEC in #1325 and to MIP, MIE, SIP, SIE in #1326.

spidugu444, 2023-08-07:
As @frikhaAziz has fixed this issue, I am closing it now.

closed 2023-08-07 by spidugu444