openhwfoundation/cva6 #1083: error when make verilate

Gorgeous0215, 2023-03-02:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

PINMISSING: ariane_testharness.sv:230:5 cell has missing pin: 'master_r_err_i'
ariane_testharness.sv:230:5   cell has missing pin : 'master_r_other_err_i

MikeOpenHWGroup, 2023-03-02:
Hi @Gorgeous0215, thanks for your interest in CVA6.  In order to help you, please provide us with some additional information.  Specifically:
- the _exact_ command-lines you invoke, starting with `git clone...`.
- information about the platform you are running on.
- the compiler/simulator and version you are using.

MikeOpenHWGroup, 2023-03-14:
Hi @Gorgeous0215, there has been no update to this issue in two weeks.  Please comment on this issue by 2023-03-21 or I will close this issue.

Gorgeous0215, 2023-03-16:
i use ubuntu20.04, following are my steps
git clone https://github.com/openhwgroup/cva6.git
git submodule update --init --recursive
and my verilator version is 4.210
when i make verilate in cva6, it came this problem:
[image]

MikeOpenHWGroup, 2023-03-23:
Thanks for the info @Gorgeous0215

> my verilator version is 4.210

AFAIK, CVA6 does not run properly with Verilator v.2xx.  We _should_ be explicit about which Verilator version to use.  V4.100 works for me and we are working on porting CVA6 to v5.008.  Please give 4.100 a try and let me know if that works for you (do not forget to delete the v4.210 generated outputs).