openhwfoundation/cva6 #1084: to get VCD file

hhhsiang, 2023-03-03:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I want to generate a VCD file to view the waveform using GTKwave to see each signal. To do this, I had executed `make verilate DEBUG=1` in advance, and I tried using the following commands under the 'cva6' folder to get a VCD file:
`work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf`
`work-ver/Variane_testharness --v $RISCV/riscv64-unknown-elf/bin/pk hello.elf`
However, both commands only displayed 'Hello CVA6!' and didn't generate a VCD file. 

I also tried the following commands from #282:
`work-ver/Variane_testharness -v dump.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf`
`work-ver/Variane_testharness -v $RISCV/riscv64-unknown-elf/bin/pk hello.elf`
The former resulted in an 'invalid option' error, while the latter didn't work. When I ran this command:
`work-ver/Variane_testharness --v dump.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf`
It displayed the message：
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 39617
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

What command should I use to generate the VCD file or should I use another tool to view the waveform? I'm new to this world, so any help would be greatly appreciated. Thank you!"

hhhsiang, 2023-03-07:
hi @sifferman, I tried the command you supply, and it seems to be useful. However, it showed error after running for a while.
These are the error message：

cd work-ver && make -j -f Variane_testharness.mk
make[1]: Entering directory '/home/aislab42190/cva6/work-ver'
Variane_testharness.mk:68: warning: overriding recipe for target 'verilated_vcd_c.o'
/usr/share/verilator/include/verilated.mk:241: warning: ignoring old recipe for target 'verilated_vcd_c.o'
make[1]: *** No rule to make target '/include/verilated_vcd_c.cpp', needed by 'verilated_vcd_c.o'.  Stop.
make[1]: Leaving directory '/home/aislab42190/cva6/work-ver'
make: *** [Makefile:639: verilate] Error 2

Do you know what's wrong with it? Thanks for your help!

hhhsiang, 2023-03-09:
@sifferman ,it can work well, thank you very much!
However, I have encountered another problem. I created a new folder named 'verilator_tool' under /home/aislab42190/ to install Verilator. I then set the environment variable using the following command:
`export VERILATOR_ROOT=/home/aislab42190/verilator_tool/`
`export PATH=$PATH:$VERILATOR_ROOT`
`source ~/.bashrc`

After that, I ran `./ci/install-verilator.sh` to install Verilator, but I noticed that Verilator was initially installed in /cva6/tmp/ and then installed to the 'verilator_tool' folder. I think this is because of the commands in 'install-verilator.sh':
[image]

I am not sure which version of Verilator is being used now - the one under /home/aislab42190/verilator_tool/ or the one under /cva6/tmp/? If I delete the cva6 directory and git clone a new one, can I still use the Verilator under /home/aislab42190/verilator_tool/? Or do I have to install it again?

sifferman, 2023-03-09:
I believe now, you want to run `make install` in your `cva6/tmp/verilator_tool` directory.

https://github.com/verilator/verilator/blob/17b75c6428a7e06961d24784ade3e47611838983/Makefile.in#L19

hhhsiang, 2023-03-10:
@sifferman, after I executed `./ci/install-verilator.sh`, I use `work-ver/Variane_testharness -v hello.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf`, and I got the file 'hello.vcd'. Why should I run `make install`? I thought that I had installed it.

sifferman, 2023-03-10:
Make install just moves all the files you built to your `$VERILATOR_ROOT`. It looks like you are all set up now, is this correct?

hhhsiang, 2023-03-12:
@sifferman, it can now successfully produce a VCD file, but the waveform looks strange. When I opened hello.vcd with GtkWave, almost all of the signals are either always high or always low, except for the "clk" signal. I'm not sure if the large size of the VCD file is causing the problem or if there are other reasons.
[image]

[image]

sifferman, 2023-03-12:
Try zooming out

hhhsiang, 2023-03-13:
@sifferman, I tried zooming out to the maximum, but it remained the same. I also noticed that the simulation time stopped at 359999ps, while I ran the simulation for at least 1800000ms according to the terminal.
[image]
[image]

Moschn, 2023-03-13:
You are looking at signals that do not change. Try looking at some other internal signals.

Wall clock time is not simulation time. Wall clock time is how long your host spent simulating the design. You can see the number of cycles simulated in the screenshot: 'completed after 2579624'.

As you managed to get a CVD file, can we close this issue?

hhhsiang, 2023-03-14:
@Moschn, OK!

JeanRochCoulon, 2023-03-14:
Hello @hhhsiang    As you opened the issue, you are the assigned person to close the issue when you think it is over (I think it is!).

hhhsiang, 2023-10-30:
@hhhsiang

closed 2023-10-30 by hhhsiang