openhwfoundation/cva6 #1091: [BUG] Under some microarchitectural circumstances, NAN conversion from simple to double precision gives wrong result

flaviens, 2023-03-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi there!

I've detected a bug in CVA6, potentially in CVFPU but I'm not certain so I post it here. Please let me know if this is a CVFPU bug, and I can move it there.

## Brief bug description

A conversion of nan from simple to double may set a lot of bits in the mantissa, unexpectedly, under specific microarchitectural conditions.

## Example instance

In this instance, we convert the single-precision `0xffffffffff800000` using `fcvt.d.s`.
We expect to get `0xfff0000000000000` as a result, but under the circumstances induced in the very simple ELF, we get `0xffefffffffffffff`. I confirmed the expected result with spike, and in the snippet further below in CVA6.

[Here is the ELF and waveforms of the bug.](https://github.com/openhwgroup/cva6/files/10891811/nancvt.zip)
A symptom of the bug is the first write to address `0x18`, which essentially writes the result of the conversion.

Here you can observe that the value dumped is erroneous.

[image]

## Example instance that runs correctly

Interestingly, performing the conversion out of context works, for example this snippet executes apparently correctly.

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:

  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  la t0, .fdata0
  fld fa5, (t0)

  fcvt.d.s ft0, fa5

  li t0, 0x18
  fsd ft0, (t0)

  sw x0, 0(x0)

infinite_loop:
  j infinite_loop

.section ".fdata0","ax",@progbits
  .8byte 0xffffffffff800000
```

Therefore I deduce that this is conditioned by some microarchitectural effects (maybe branch prediction but I am not certain).

Thanks!
Flavien

MikeOpenHWGroup, 2023-03-06:
Hi @flaviens, thanks for this issue!  Can you close this issue and re-post it to the [CVFPU repo](https://github.com/openhwgroup/cvfpu)?  As you may know, the CVA6 team is working to verify v5.0.0 which does not support the FD ISA.  Of course a future release of CVA6 _will_ support FD and will use the CVFPU for that.

It is important to get this issue in the CVFPU repo as the CV32E40P v2.0.0 integrates CVFU and that project will need a fully verified floating point unit much eariler than CVA6.

flaviens, 2023-03-06:
Hi @MikeOpenHWGroup , thank you for your response, I'll do this! I initially decided to post here because the bug symptom _may_ result from a CVA6-specific microarchitectural condition. Will keep it in mind in case another design could not reproduce these specific instance.

closed 2023-03-06 by flaviens