openhwfoundation/cva6 #1095: PMP Granularity needs to be discussed in User Manual

MikeOpenHWGroup, 2023-03-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The PMP granularity of CVA6 is fixed at 8.  This is fine, but as PMP granularity is visible to the user (e.g. software developers), this information needs to be available in the [PMP Chapter of the User Guide](https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/PMP.html).  Currently it is only specified in the Design Specification of the [Execution P Stage](https://docs.openhwgroup.org/projects/cva6-user-manual/03_cva6_design/ex_stage.html#pma-pmp-checks).

Moschn, 2023-03-06:
A small comment: The PMP granularity _is_ visible to the user. The RISC-V priviledge spec states: 

> Software may determine the PMP granularity by writing zero to pmp0cfg, then writing all ones to pmpaddr0, then reading back pmpaddr0. If G is the index of the least-significant bit set, the PMP granularity is 2G+2 bytes.

zarubaf, 2023-03-06:
All added in #1092

PR #1092 (2023-03-10): Add PMP section and fix addressing CV64A6
As requested here is the PMP section for the user guide, plus a fix for the CSRs for CV64A6 which should trap on PMP access `CSR_PMPCFG1` and `CSR_PMPCFG3`. 

Fixes #1095
closed 2023-03-10 by JeanRochCoulon