openhwfoundation/cva6 #1096: Performance Counter CSRs in User Manual do not align with performance counters

MikeOpenHWGroup, 2023-03-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The [Performance Counters section of the Requirements Specification](https://docs.openhwgroup.org/projects/cva6-user-manual/02_cva6_requirements/cva6_requirements_specification.html#performance-counters) indicate that there should be six 64-bit performance counters (`HPM-20`) and these performance counters shall be selected by the mhpmevent3 to mhpmevent8 CSRs  (`HPM-40`).  This is contradicted by the User Manual which implies that specific performance counter functions are hardwired to specific CSRS (e.g. [L1-Icache-miss](https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/CV32A6_Control_Status_Registers.html#l1-inst-cache-miss-ml1-icache-miss)).

JeanRochCoulon, 2023-03-06:
@TulikaSi    COuld you help us on this specification and implementation point ?

TulikaSi, 2023-03-07:
Hi @JeanRochCoulon and @MikeOpenHWGroup 
As per the RISC V Specification (**Section 3.1.10 of Volume 2, Privileged Spec v. 20211203** ), the hardware performance monitor includes 29 no. of 64-bit event counters, **mhpmcounter3– mhpmcounter31** along with the event selector CSRs, **mhpmevent3–mhpmevent31** and the same has been mentioned in the Performance Counter Section of the Requirement Specification. An issue [https://github.com/openhwgroup/cva6/issues/683]had been raised regarding the length of the performance counters and a PR has been submitted in order to comply with the length as well the implementation as per the RISCV Specification -----[https://github.com/openhwgroup/cva6/pull/1034] which can be reviewed(regarding the implementation part)but few modifications are required before merging it.

TulikaSi, 2023-12-11:
Hi @MikeOpenHWGroup @JeanRochCoulon 
The document ( https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CSR_Performance_Counters.rst ) clearly explains about the perf counters but the **CV32A6_Control_status_Register_rst doc** ( https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CV32A6_Control_Status_Registers.rst ) needs to be modified to comply with the implementation so that it does not create a confusion.