openhwfoundation/cva6 #1100: illegal instruction exception handler in CVA6

spidugu444, 2023-03-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

While verifying access modes in CVA6, while trying to attempt write on RO(Read Only) CSR's causes an illegal instruction exception and test has been ended with WFI.
 Is there any way to verify the situation where a illegal instruction exception is expected.

Moschn, 2023-03-13:
This is not necessarily a CVA6 question, but take a look how [riscv-tests](https://github.com/riscv-software-src/riscv-tests) check for instructions that are supposed to throw an exception. For example, the PMP benchmark first sets up a trap handler that checks if a trap was expected (https://github.com/riscv-software-src/riscv-tests/blob/master/benchmarks/pmp/pmp.c#L15). Another example of a trap handler to check if an exception was expected can be found in `env` in risc-tests: https://github.com/riscv/riscv-test-env/blob/7283c38f3a56d8fa8d95733cad015ac587fa9277/p/riscv_test.h#L171.

JeanRochCoulon, 2023-03-14:
Thank @Moschn, did this suggestion answered your question @spidugu444 ?  If yes, you can close the issue, if no, you can re-assign the issue.

spidugu444, 2023-07-24:
@Moschn, Thanks for your inputs.

Please find the attached Machine mode read-only CSR testcase where I can access CSRs in all privileged modes (Machine supervisor and user mode) and control exceptions using an exception handler.
I Am attaching log and csv files for the same, please let me know if any changes needed.

Machine read-only test:
[riscv_m_ro_csr_test.txt](https://github.com/openhwgroup/cva6/files/12142625/riscv_m_ro_csr_test.txt)

log and csv files:
[riscv_spike_m_ro_csr_test_0.csv](https://github.com/openhwgroup/cva6/files/12142607/riscv_spike_m_ro_csr_test_0.csv)
[riscv_spike_m_ro_csr_test_0.log](https://github.com/openhwgroup/cva6/files/12142608/riscv_spike_m_ro_csr_test_0.log)
[riscv_vcs_m_ro_csr_test_0.csv](https://github.com/openhwgroup/cva6/files/12142609/riscv_vcs_m_ro_csr_test_0.csv)
[riscv_vcs_m_ro_csr_test_0.log](https://github.com/openhwgroup/cva6/files/12142610/riscv_vcs_m_ro_csr_test_0.log)

Thanks

Moschn, 2023-07-24:
What are these logs related to? This issue is about illegal instruction handlers.

RanjanThales, 2023-08-11:
Hi @Moschn , what @spidugu444 is trying to put here is that -> his test scenario intent to write some random data into RO (Read only) CSRs, which is expected to illegal exception (part of csr access verification). Then the logic that we were looking into how to tackle that exception and continue the program (till end !). And I believe your previous pointers and inputs from @MikeOpenHWGroup helped us to achieve it. https://github.com/openhwgroup/core-v-verif/pull/2105 (the previous logs are also part of it)
I believe, we can close these issue now, @spidugu444 !!

spidugu444, 2023-08-14:
Yes, @RanjanThales, you're right. I raised this issue on how to handle that excepted exceptions and keep running the program to the very end, when we write data into RO CSR, and I was successful in doing so.
As stated in the comment above, I included RO CSRs access verification test program where I am handling the expected exceptions and log files has been attached for the same.
As I achieved my requirement, I am closing this issue.

closed 2023-08-14 by spidugu444