openhwfoundation/cva6 #1106: [BUG] Out-of-Order Not Working

sifferman, 2023-03-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

## Bug Explanation

I noticed the docs say that Out-of-Order can occur between the multiplier and the ALU, however this is not the case according to the RTL.

Docs:

https://github.com/openhwgroup/cva6/blob/a29b231070f2dccca53eae2681120116633a4ffc/docs/03_cva6_design/issue_stage.md?plain=1#L42-L54

RTL:

https://github.com/openhwgroup/cva6/blob/adf99b53042d2b14cbbb36689babf8cd4c03cf81/core/issue_read_operands.sv#L135-L152

For example, I have this code snippet:

```
        div     t2, t0, t1
        fsqrt.s f1, f1
        lw      t5, 0(s1)
        add     t6, t0, t1
```

And this is it's execute stage:

[image]

As you can see, the `add` instruction isn't run until after the `div` finishes.

## Proposed Fix

Either:

* The RTL could be changed to separate the ALU and multiplier into two separate busy signals.
* The example in the docs could be changed to something else, such as `div` and `lsu`.

sifferman, 2023-03-10:
If you like, I can submit a pull request changing the docs to give div and lsu as the example

cathales, 2023-05-11:
Hello @sifferman and thanks for opening this issue.

I had a look, I think I have found a too strict restriction (#1230) related to `mult -> *` sequences. However it does not seem to fix this issue about `div -> flu` sequences.

The parallelism is only possible for units spread on different result buses, so that their result can be provided to the scoreboard without conflicts. For the `div -> add` sequence you show, they use the same result bus (named `flu` as "fixed latency unit", even if div does not have a fixed latency), so you are right, the doc example is wrong.

A few suggestions/ideas:

- We can update the docs as you suggest.
- We could add a write-back port for `mult/div` but we won't as the area cost would be considerable!
- We can check if there is a minimal latency for div; if so, maybe we could issue a few other FLU instructions just after the `div` starts?
- Maybe we could modify how the div result is handled: when div result is available, if result bus is available then push the result, else store it in a (new?) register and lock the `flu` issues until the result bus is available, then release the lock and push the result.
- Maybe we could move the `mult/div` unit to the same port as FP operations?

sifferman, 2023-05-11:
Thanks for taking a look @cathales! I agree with your observations.

Actually, I believe that the throughput could increase quite a bit if the mult/div unit uses the FP port. On a mult/div instruction, it's probably more likely that the next instruction will be an ALU instruction than a FPU instruction. But I'm not sure how big of an endeavor it would be to complete this.

I will have time to look into it in about a month if progress hasn't been made on it sooner

cathales, 2023-05-12:
If we move mult/div to FP port, I hope multiply-accumulate would not need bubbles anymore if the compiler is smart enough to avoid RAW hasards like below:

```asm
mul m0, a, b
; begin repeat
mul m1, a, b ; issue ok
add r, r, m0 ; now: bubble because same bus than previous mul -> after modification: issue ok
mul m0, a, b ; issue ok
add r, r, m1 ; bubble -> issue ok
; end repeat
add r, r, m0 ; issue ok
```

However, if the compiler does not do that (see below), there will be the same bubbles anyway so the modification will not impact performance.

```asm
; begin repeat
mul m, a, b ; issue ok
add r, r, m ; bubble because needs m from mul above and mul duration is 2 cycles
; end repeat
````

cathales, 2023-05-31:
> Actually, I believe that the throughput could increase quite a bit if the mult/div unit uses the FP port. On a mult/div instruction, it's probably more likely that the next instruction will be an [...]

For CoreMark the increase would be less than 3%, but it depends of the application.

About area:

For a configuration with both F and M, we use an existing write-back port so it is okay.

For a configuration with M, without F, today there should not be a write-back port for FPU, so the muxes in the scoreboard could be cheap. If we add one port for the multiplier, it means that the muxes are big.

So maybe the performance vs area cost should be discussed.

PR #1621 (2023-11-13): docs: fix example of OoO write-back to scoreboard
Multiplication takes two cycles (1 cycle latency) and ALU takes one
cycle (no latency). They share the same writeback port so it is not
possible to issue an ALU instruction just after a MUL.  So the example
is wrong, but it is okay if we replace MUL by LOAD as it uses another
write-back port.

Fix #1106
commit 758511fd (2023-11-13): docs: fix example of OoO write-back to scoreboard (#1621)

Multiplication takes two cycles (1 cycle latency) and ALU takes one
cycle (no latency). They share the same writeback port so it is not
possible to issue an ALU instruction just after a MUL.  So the example
is wrong, but it is okay if we replace MUL by LOAD as it uses another
write-back port.

Fix #1106
closed 2023-11-13 by JeanRochCoulon