openhwfoundation/cva6 #1109: [BUG] Dependency exists between module input parameters

JeanRochCoulon, 2023-03-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

@Nils Wistoff @Florian Zaruba,   In the ARA PR, AXI parameters are defined as seen below. I think this code cannot work. As example, when AxiAddrWidth is defined by ariane module (which instantiates cva6) and axi_ar_chan_t is not defined by ariane module, axi_ar_chan_t will be defined from ariane_axi::AddrWidth, which is not the wanted value. Because in that case, axi_ar_chan_t should be defined from AxiAddrWidth.  Tell me if I am wrong !
Moreover, almost all the parameters are defined in config package, why defining them in the module parameter ?   Code would be simpler by defining parameters always with the same strategy. If you need a special configuration for ARA, what I can understand of course, you can add an ARA configuration to the cva6 repository.
[image]

niwis, 2023-03-09:
Hi @JeanRochCoulon, this is actually not in the Ara PR (anymore), but it was merged into master in #982 and #1103

I don't see the bug. Both `AxiAddrWidth` and `axi_ar_chan_t` are passed from `ariane` to `cva6` [here](https://github.com/openhwgroup/cva6/blob/fdc7216920f6d8ea2474b9ecca8b82da86295814/core/ariane.sv#L62-L71), or am I missing something? Of course, we will need to rebase the Ara branch onto `master` now that those changes have been moved to separate PRs.

`axi_ax_chan_t` depends not only on the address width but also on the id and user width. But I think it should also be possible to pass these values and re-define the channel types within `std_subsystem` where they are used. I think this is a matter  of preference.

Are you suggesting moving these parameters to `ArianeCfg` or `cva6_config_pkg`? I think the latter would make it more difficult to pass these parameters from a system that instantiates CVA6.

PS: Not sure how interested Florian Hartmann and Nils Schnabel are in this discussion :-)

JeanRochCoulon, 2023-03-09:
Hi,
The bug occurs when defining AxiAddrWidth without defining axi_ar_chan_t.
CVA6 is configured by cva6_config_pkg: Jonathan Balkin added https://github.com/openhwgroup/cva6/blob/master/core/include/cv64a6_imafdc_sv39_openpiton_config_pkg.sv for instance. I suggest you to create a dedicated cva6_HW_config for ARA and define AXI paramaters inside: accelarator port ? FPU SIMD ? scoreboard entry number? extension A? extension C? ....

zarubaf, 2023-03-09:
> The bug occurs when defining AxiAddrWidth without defining axi_ar_chan_t

That is not a legal configuration. Either you rely on the package or you need to overwrite both.

JeanRochCoulon, 2023-03-14:
By defining a structure, as done for arianecfg, the user would be constraint to define all the members of the structure. That's a way to avoid illegal configuration. And it would generate a code more compact !

zarubaf, 2023-10-23:
This is fixed!

closed 2023-10-23 by zarubaf