openhwfoundation/cva6 #1116: minstreth CSR is failing while verifying access modes

spidugu444, 2023-03-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

As per riscv priviliged specification **minstreth CSR return bits 63–32 of machine instructions retired counter**.

While trying to verify access modes for minstreth CSR test is failing on verilator end as it is always reading Zero.
whereas on spike end test is reading expexted values and test is passing.
For reference please find attached spike and verilator log files

[riscv_spike_minstreth_csr_test_0.log](https://github.com/openhwgroup/cva6/files/10976594/riscv_spike_minstreth_csr_test_0.log)
[riscv_verilator_minstreth_csr_test_0.log](https://github.com/openhwgroup/cva6/files/10976599/riscv_verilator_minstreth_csr_test_0.log)

JeanRochCoulon, 2023-03-15:
Thanks @spidugu444 for having reported this issue. The concern is that so far no designer has been identified to fix the RTL. The main thing is the fact that the issue is tracked.

spidugu444, 2023-03-15:
Hi @Moschn, with your fix test is working fine, Thanks for your fix.
I am closing this issue.

closed 2023-03-15 by spidugu444

JeanRochCoulon, 2023-03-15:
@spidugu444  do you have double checked that the PR fixes the bug before closing the Issue ?

spidugu444, 2023-03-16:
HI @JeanRochCoulon, before closing the issue I confirmed it by running arch-test suite regression and minstreth CSR access mode test.
please find the attached minstreth CSV files for your reference.

[riscv_spike_minstreth_csr_test_0.csv](https://github.com/openhwgroup/cva6/files/10987385/riscv_spike_minstreth_csr_test_0.csv)
[riscv_verilator_minstreth_csr_test_0.csv](https://github.com/openhwgroup/cva6/files/10987386/riscv_verilator_minstreth_csr_test_0.csv)

JeanRochCoulon, 2023-03-16:
Great !   Thanks Sai, have a nice day

PR #1117 (2023-03-15): Fix #1116
I did not test this myself, but I think it should fix #1116. @spidugu444 could you test this patch?