openhwfoundation/cva6 #1122: [BUG] std_cache_subsystem (write-back cache) cannot be currently used

cfuguet, 2023-03-17:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,

I've noticed that currently the write-back cache cannot be used with the current compilation makefiles and list of files for the CVA6.

There are two issues:
1. In the Flist.cva6 there is a '+define+WT_DCACHE' that forces the use of the write-through cache. This define shall be removed.
2. std_cache_subsystem files are not listed.

In the cva6.sv file, there is a '#ifdef WT_DCACHE .... #else '. The only way to use the write-back cache is to not define the WT_DCACHE macro. However, as mentioned above, it is not currently possible.

By the way, this issue means that the CI, that currently supposes to execute tests on both WT and WB caches, do not actually do it.

JeanRochCoulon, 2023-03-17:
Define are so dangerous !!!   We must remove them !

JeanRochCoulon, 2023-03-21:
@cfuguet  #1127 should fix the bug. Can you double check ?

cfuguet, 2023-03-21:
Yes, that is an option. The drawback is that we will have code replication in every config file. But not that significant I guess.

cfuguet, 2023-11-09:
This was already fixed

closed 2023-11-09 by cfuguet