openhwfoundation/cva6 #1126: [BUG] UVM Failing at instruction page fault

SylRemTHA, 2023-03-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I'm currently trying to update the cva6 from commit 75250868eb5da568ee86be26b2fb660fc245a771 to the latest version. 
The previous version was working without any problem and showed "Hello cva6!" when simulated with questasim as intended.
The latest version fails at exception "invalid instruction fetch". The "invalid instruction fetch" also occured in the previous version but didn't cause the entire simulation to fail somehow.

I've also noticed some change in values in the trace_hart_0 that I couldn't manage to understand.
I've attached both working and non working trace_hart_0 files as well as the non working transcript. [logs.zip](https://github.com/openhwgroup/cva6/files/11018832/logs.zip)
You will notice that they diverge at `43870010ns`, `43870030ns` without impacting the simulation. 
But then starting from the instruction at `58188610ns` (working file) and `58187610ns` (non working file), the simulations diverge completely.

Both versions were compiled and run the same way and no personnal modifications have occured.

Thank you for your assistance.

zchamski, 2023-03-23:
Hi @SylRemTHA , to help with the analysis of the issue, can you indicate the CVA6 hash that you used in the failing case?

MikeOpenHWGroup, 2023-03-23:
Also, @SylRemTHA, please update this issue with the exact command-lines you use to create this issue, starting with `got clone ...`.  Attaching a logfile would also be helpful.  Thanks!

SylRemTHA, 2023-03-24:
hi @zchamski, the latest version I was trying to use was hash 2c61865b1856ad038f66e1fef62501daf855806c.
For the command lines, for both versions, I've used : `git clone https://github.com/openhwgroup/cva6.git` and `git submodule update --init --recursive`. I then launched `./ci/setup.sh` and everything installed smoothly, at least there are no signs of failing anywhere on that part.

zchamski, 2023-12-12:
Hi @SylRemTHA, any updates?

JeanRochCoulon, 2025-01-02:
This issue is inactive for a while. Please open another if needed.

closed 2025-01-02 by JeanRochCoulon