openhwfoundation/cva6 #1131: mstatus CSR is failing while verifying access mode.

spidugu444, 2023-03-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

While verifying access mode for MSTATUS CSR, Test got failed on RTL end due to reading nonzero values on reserved fields, while on spike end reserved fields are reading with zero.

please find the steps to replicate issue.

1. git clone https://github.com/spidugu444/core-v-verif.git
2. git check out csr_mstatus_access_verification
3. set toolchain to RISCV varible
4. Run MSTATUS test by using **source cva6/regress/smoke-tests.sh**
5. output directory will generate inside cva6/sim 
6. MSTATUS test and Yaml file location:  cva6/tests/custom/csr_access_tests

JeanRochCoulon, 2023-06-08:
Hello @spidugu444,
@frikhaAziz has just fixed the issue with PR #1257. Can you double check and close the issue if it is the case ?

spidugu444, 2023-06-14:
HI @JeanRochCoulon, 

with the updated CVA6 I am reading RTL value for MSTATUS CSR as 0x007E1FEA, previously it is 0x7FFE1FEE.

With updated RTL reserved fields in bit [2], bit [30:23] is reading zeros which is working fine.

In CVA6 user Manual bit [6], bit[10:9] fields named as reserved but in riscv privileged 20211203 specification those fields are marked as UBE and VS with access type as WARL which is as shown below.

I believe cva6-user-manual and riscv privileged 20211203 specification has not compliance here.

cva6-user-manual:
[image]

 riscv-privileged-20211203 specification
[image]

spidugu444, 2023-07-05:
Hello, @frikhaAziz. As per discussion with @ASintzoff in MSTATUS CSR bit[6] UBE and bit[10:9]VS fields should be reserved I am reading nonzero values in UBE and VS fields, which are intended to be reserved as per the cva6 user manual.

spidugu444, 2023-07-05:
@RanjanThales

frikhaAziz, 2023-07-20:
@MikeOpenHWGroup @spidugu444 I just added the fields VS and UBE in #1308 to make it compliant to risc-v privileged 20211203.

MikeOpenHWGroup, 2023-08-23:
Hi @spidugu444, please re-test this CSR access mode again against the latest RTL.

spidugu444, 2023-08-28:
With @frikhaAziz  fixes on RTL end test is working fine, but on spike end there are some discrepancies for FS and UBE field. 
I am closing this issue as I addressed spike discrepancies in this [issue](https://github.com/openhwgroup/cva6/issues/1346).

closed 2023-08-28 by spidugu444