openhwfoundation/cva6 #1133: make cva6 synthesis

michael116688, 2023-03-23:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I'm trying to perform cva6 synthesis and encountered an error while running the following command:

`make -C pd/synth cva6_synth FOUNDRY_PATH=/your/techno/basepath/ TECH_NAME=yourTechnoName TARGET_LIBRARY_FILES="yourLib1.db\ yourLib2.db" PERIOD=10 NAND2_AREA=650 TARGET=cv64a6_imafdc_sv39 ADDITIONAL_SEARCH_PATH="others/libs/paths/one\ others/libs/paths/two"`

The error message I received is:

> make: Entering directory '/home/ais04/cva6/pd/synth'
grep "CVA6_REPO_DIR\}" ../../core/Flist.cva6|grep -v "instr_tracer"|grep -v "incdir" > Flist.cva6_synth
sed -i "s/^/analyze -f sverilog -define {WT_DCACHE,RVFI_TRACE,RVFI_MEM} -lib ariane_lib /" Flist.cva6_synth
10
/bin/sh: 1: /opt/synopsys/syn/Q-2019.12/bin//dc_shell: not found
make: *** [Makefile:37: cva6_synth] Error 127
make: Leaving directory '/home/ais04/cva6/pd/synth'

I'm not sure what went wrong. Do I need to install any additional tools to fix this issue?

Thank you.

MikeOpenHWGroup, 2023-03-23:
Is _your_ path to **dc_shell**  `/opt/synopsys/syn/Q-2019.12/bin//dc_shell`  Do you have `dc_shell` installed?

michael116688, 2023-03-27:
@MikeOpenHWGroup
I add the path as follows:
`export PATH=/opt/synopsys/syn/Q-2019.12/bin//dc_shell:$PATH`
but, I am still getting the same error message.
I'm not sure if `dc_shell` is installed or not. How can I check for this?
Thank you.

JeanRochCoulon, 2023-10-24:
Related to  #1569

valentinThomazic, 2023-12-07:
Hey @michael116688, did you manage to solve your issue in the meantime ?

michael116688, 2023-12-07:
> Hey @michael116688, did you manage to solve your issue in the meantime ?

Yes, I managed my issue.

valentinThomazic, 2023-12-08:
Great, could you close it then ?