openhwfoundation/cva6 #1144: [TASK] <LINT: Duplicate declaration of module:axi_adapter>

umerimran-10xe, 2023-03-27:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

**Environment:**

Operating System: [_Ubuntu 20.04_]
Compiler Version: [_gcc 9.4.0_]
CVA6 Version: [_SHA:7f4549f2_]

When compiling CVA6 with Verilator 5.008, a lint warning is generated due to the Duplicate declaration of module. Specifically, the same module is being declared in multiple files, which can lead to confusion.  Is it intentional or should we try to resolve this warning ?

```
%Warning-MODDUP: cva6/corev_apu/tb/axi_adapter.sv:19:8: Duplicate declaration of module: 'axi_adapter'
   19 | module axi_adapter #(
      |        ^~~~~~~~~~~
                 cva6/core/cache_subsystem/axi_adapter.sv:19:8: ... Location of original declaration
   19 | module axi_adapter #(
      |        ^~~~~~~~~~~
                 ... For warning description see https://verilator.org/warn/MODDUP?v=5.008
                 ... Use "/* verilator lint_off MODDUP */" and lint_on around source to disable this message.
```

### Description of Done

Lint Warning is removed or waived off.

JeanRochCoulon, 2023-03-29:
@mike, I assigned you to create the label, do not hesitate to send me back the hot potato after !

MikeOpenHWGroup, 2023-03-29:
>  Is it intentional or should we try to resolve this warning ?

I doubt very much that this is intentional.  The same file is located in two different directories.  This is a very bad thing and must be fixed.  There are several ways this can be done:
1. Rename `core/cache_subsystem/axi_adapter.sv` to `core/cache_subsystem/cva6_axi_adapter.sv` (both the file and module name must be changed).
2. Place axi_adapter.sv in a common directory.
3. Put compile guards around the module in the source-file.

umerimran-10xe, 2023-04-10:
#1174 Removed this, closing this issue

closed 2023-04-10 by umerimran-10xe