openhwfoundation/cva6 #1145: SSTATUS CSR is failing while verifying Access mode

spidugu444, 2023-03-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

While verifying Access mode for SSTATUS CSR test is failing on RTL end due to FS field in CSR is always reading Zero.

please find the steps to replicate the issue.

1. git clone https://github.com/spidugu444/core-v-verif.git
2. git checkout csr_sstatus_issue
3. set toolchain to RISCV variable
4. Run SSTATUS test by using source cva6/regress/smoke-tests.sh
5. output directory will generate inside cva6/sim
6. SSTATUS test and Yaml file location: cva6/tests/custom/csr_access_tests

For quick reference please find spike and VCS log files as attached.

[riscv_spike_sstatus_csr_test_0.log](https://github.com/openhwgroup/cva6/files/11089344/riscv_spike_sstatus_csr_test_0.log)
[riscv_vcs_sstatus_csr_test_0.log](https://github.com/openhwgroup/cva6/files/11089334/riscv_vcs_sstatus_csr_test_0.log)

spidugu444, 2023-04-11:
I found that in CVA6 RTL due to disable of Floating-point unit FS field in sstatus is always reading zero.
so we need to configure spike to disable FS field.

ASintzoff, 2023-11-14:
This issue is linked to Spike configuration. `mstatus.FS` and `sstatus.FS` field should be configurable

MarioOpenHWGroup, 2023-11-21:
Is this specified somewhere in the User Manual?

zchamski, 2023-11-21:
Two issues here, actually:
* does the configuration under test support `SSTATUS` at all, knowing that the "embedded" config only has M mode?
* [here](https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/CV32A6_Control_Status_Registers.html#supervisor-status-sstatus) the `FS` field of `STATUS` is explicitly described as `RW`.  However, the behavior of the `FS` filed in the absence of an FPU is not described.

Bottom line: the `SSTATUS` specification in the CVA6 User Manual needs to be clarified wrt. configurations and their properties, including the presence or not of the FPU in a given config.

MarioOpenHWGroup, 2024-08-26:
`FS` field of the CSR can controlled by `_write_mask` param of the tandem, closing the issue

closed 2024-08-26 by MarioOpenHWGroup