openhwfoundation/cva6 #1148: [TASK] LINT: P2 - Selection index out of bounds

umerimran-10xe, 2023-03-29:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Environment:

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 Version: [SHA:7f4549f2]

When compiling CVA6 with Verilator 5.008, a lint warning is generated due to the selection index being out of bounds. Verilator will assume zero for this value. The problem with the below mentioned code is that the User width is 1 bit, so we can probably use `ariane_pkg::DCACHE_USER_WIDTH` here. I think that the logic would remain same after that since it is using the construct `+:` part-select addressing but I want to know the actual intent here.

```
%Warning-SELRANGE: cva6/core/cache_subsystem/wt_dcache_wbuffer.sv:487:37: Extracting 8 bits from only 1 bit number
                                                                                              : ... In instance ariane_testharness.i_ariane.i_cva6.genblk3.i_cache_subsystem.i_wt_dcache.i_wt_dcache_wbuffer
  487 |               wbuffer_d[wr_ptr].user[k*8 +: 8] = req_port_i.data_wuser[k*8 +: 8];
      |                                     ^

```

### Description of Done

Lint Warning is removed or waived off.

JeanRochCoulon, 2023-04-03:
#1159 fixes this issue. If you are happy with this fix, you can close the issue. Thanks !!

umerimran-10xe, 2023-04-08:
Updating the main warnings issue with the status and closing this issue

closed 2023-04-08 by umerimran-10xe