openhwfoundation/cva6 #1149: [TASK] LINT: P3 - Combinational block order

umerimran-10xe, 2023-03-29:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

**Environment:**

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 Version: [SHA:7f4549f2]

When compiling CVA6 with Verilator 5.008, a lint warning is generated due to the fact that `always_comb` block has a variable that is set after it is used. This happens in only one case where the co-processor has a some data assigned to it before the actual change in the value.  

```
%Warning-ALWCOMBORDER: cva6/core/cvxif_example/cvxif_example_coprocessor.sv:146:5: Always_comb variable driven after use: 'x_result_o'
                                                                                                       : ... In instance ariane_testharness.i_ariane.gen_example_coprocessor.i_cvxif_coprocessor
  146 |     x_result_o.data     = req_o.req.rs[0] + req_o.req.rs[1] + ( X_NUM_RS == 3 ? req_o.req.rs[2] : 0);
      |     ^~~~~~~~~~
```

### Description of Done

Lint Warning is removed or waived off.

JeanRochCoulon, 2023-04-02:
@Gchauvon, Umer has detected an RTL syntax to be improved. Can you have a look ?

umerimran-10xe, 2023-05-05:
Any update on this one ?

closed 2023-06-26 by umerimran-10xe