openhwfoundation/cva6 #1150: [TASK] LINT: P3 - Symbol matching C++ keyword

umerimran-10xe, 2023-03-29:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

**Environment**

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 Version: [SHA:7f4549f2]

When compiling CVA6 with Verilator 5.008, a lint warning is generated due to the Symbol matching C++ keyword: 'operator'. Since verilator generates behavioral C++ Models, it's better to avoid C++ keyword names in the SV Code. We have to change this at different places within the code, what's the suggestion here ?

```
%Warning-SYMRSVDWORD: cva6/core/include/ariane_pkg.sv:551:35: Symbol matches C++ keyword: 'operator'
  551 |         fu_op                     operator;
      |                                   ^~~~~~~~
```

### Description of Done

Lint Warning is removed or waived off.

JeanRochCoulon, 2023-04-02:
@umerimran-10xe   If you think the github issue is over, you can close the issue. I assign it to you.

umerimran-10xe, 2023-04-02:
Updated the status in #1155, closing this issue

closed 2023-04-02 by umerimran-10xe