openhwfoundation/cva6 #1151: [TASK] LINT: P1 - Blocked and non-blocking assignments to same variable

umerimran-10xe, 2023-03-29:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

**Environment**

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 Version: [SHA:7f4549f2]

When compiling CVA6 with Verilator 5.008, a lint error is being generated due to the fact that Blocked and non-blocking assignments have been made to the same variable in many places in floating point unit. By default, the latest version of verilator consider this as an error instead of a warning, therefore making it as a higher priority than other Lint issues. I think there will be problems with Synthesis tools related to this particular case, so we need to come up with a solution for this one.

```
%Error-BLKANDNBLK: /home/umerimran/Music/cva6/vendor/pulp-platform/fpnew/src/fpnew_fma.sv:106:59: Unsupported: Blocked and non-blocking assignments to same variable: 'ariane_testharness.i_ariane.i_cva6.ex_stage_i.fpu_gen.fpu_i.fpu_gen.i_fpnew_bulk.gen_operation_groups[0].i_opgroup_block.gen_parallel_slices[0].active_format.i_fmt_slice.gen_num_lanes[0].active_lane.lane_instance.i_fma.inp_pipe_valid_q'
  106 |   logic                  [0:NUM_INP_REGS]                 inp_pipe_valid_q;
      |                                                           ^~~~~~~~~~~~~~~~
                   /home/umerimran/Music/cva6/vendor/pulp-platform/fpnew/src/fpnew_fma.sv:118:33: ... Location of blocking assignment
  118 |   assign inp_pipe_valid_q[0]    = in_valid_i;
      |                                 ^
                   /home/umerimran/Music/cva6/vendor/pulp-platform/fpnew/src/fpnew_fma.sv:130:29: ... Location of nonblocking assignment
  130 |       inp_pipe_valid_q[i+1] <= (flush_i) ? (1'b0) : (inp_pipe_ready[i]) ? (inp_pipe_valid_q[i]) : (inp_pipe_valid_q[i+1]);       
      |                             ^~
```

### Description of Done

Lint Warning should be removed.

JeanRochCoulon, 2023-04-02:
This is indeed a critical Lint warning/error. As FPnew is being updated, fixing it Today seems useless. This Github issue could be replayed on the updated fpnew. One or two weeks would be needed to have the new fpnew.

umerimran-10xe, 2023-04-10:
Status update: waiting for #1102 to merge with master

JeanRochCoulon, 2023-04-16:
Hello @umerimran-10xe,   good news, the PFU has just been updated.  Can I ask you to verify that the LINT problem you saw has been fixed and close the issue if it is the case ?

umerimran-10xe, 2023-04-17:
Sure, let me see, Thanks :)

closed 2023-04-17 by umerimran-10xe