openhwfoundation/cva6 #1155: [TASK] Focal Point of all CVA6 RTL Lint warnings

umerimran-10xe, 2023-04-02:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

## Introduction
### Environment

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 version: [SHA:7f4549f2]
Verilator version: [5.008] 

### Background
RTL Linting is used to verify adherence to specified design coding rules that prevent synthesis issues, functional bugs and enforce coding styles for readability & re-use. For that purpose, we have to enable lint warnings and remove/waive them where necessary. To achieve this objective, a set of PRs will be created removing a batch of lint warnings. This issue tracks the status updates and the relative PRs related to Lint warnings related to all the github lint cases.     

### Guidelines for Lint Issues

- All the CVA6 Lint Related issues will start with LINT keyword and provide the specific info related to the warning
- Priorities are being assigned to every Lint Issue with P1 being the highest
  - Resolution of P1 Lint Issues -> 7 days
  - Resolution of P2 and P3 Lint Issues -> 15 days
- Deferred (represented by D) Issues are those which are outside core/ directory and don't have much priority right now

## Status 

This table specifies all the warnings related to lint in CVA6 RTL and classifies those into specific categories. The progress of all the Lint Issues will be maintained here.

| Lint Warning | Priority | Issue | Status | PR | 
| :-----------: | :-----------: | :-----------: | :-----------: | :-----------: |
| MODDUP | P1 | #1144 | In Progress | - |
| ALWCOMBORDER | P3 | #1149  | In Progress | - |
| SYMRSVDWORD | P3 | #1150 | In Progress | #1154 |
| SELRANGE | P2 | #1148 | In Progress | - |
| WIDTHEXPAND | P2 | - | In Progress | - |
| WIDTHCONCAT | P2 | - | In Progress | - |
| WIDTHTRUNC | P2 | - | In Progress | - |
| STMTDLY | P2 | - | In Progress | - |
| CASEINCOMPLETE | P1 | - | In Progress | - |
| DECLFILENAME | - | - | In Progress | - |
| LATCH | - | - | None | - |
| UNOPTFLAT | - | - | None | - |
| IMPORTSTAR | - | - | None | - |
| VARHIDDEN | - | - | None | - |
| UNUSEDSIGNAL | - | - | None | - |
| UNUSEDPARAM | - | - | None | - |
| UNDRIVEN | - | - | None | - |
| SYNCASYNCNET | - | - | None | - |
| BLKSEQ | - | - | None | - |
| PINCONNECTEMPTY | - | - | None | - |
| UNSIGNED | - | - | None | - |
| LITENDIAN | D | - | None | - |
| BLKANDNBLK | D | #1151 | None | - |
| CMPCONST | D | - | None | - |

Since Mike has created a Lint label for these type of issues, so a table depicting overall status of this task here :slightly_smiling_face: 
cc/ @MikeOpenHWGroup @ASintzoff @JeanRochCoulon 

### Description of Done

- [ ] Resolution of `cva6/core/` directory lint problems
- [ ] All the related lint waivers are in place
- [ ] Issues for all deferred items has been created

JeanRochCoulon, 2023-04-14:
Hello Umer,   The current verification focused on step1 with is called CV32A60X. But the Thales CI simulates also CV64A6 configuration. That's why LINT should be fixed for CV32A60X, but the resulting RTL should run on CV32A60X and CV64A6.

closed 2023-09-15 by fatimasaleem

umerimran-10xe, 2023-09-15:
#1377 will be taking care of all the lint warnings going forward.