openhwfoundation/cva6 #1157: [TASK] LINT: P2 - Circular Combinational Logic

umerimran-10xe, 2023-04-03:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

**Environment**

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 Version: [SHA:7f4549f2]

When compiling CVA6 with Verilator 5.008, a lint error is being generated due to circular combinational logic in `core/` directory. We need to see these warnings and deal with the code that is generating this logic.

```
%Warning-UNOPTFLAT: /home/umerimran/Music/cva6/core/cva6.sv:248:29: Signal unoptimizable: Circular combinational logic: 'ariane_testharness.i_ariane.i_cva6.icache_dreq_if_cache'
  248 |   icache_dreq_i_t           icache_dreq_if_cache;
      |                             ^~~~~~~~~~~~~~~~~~~~
                    /home/umerimran/Music/cva6/core/cva6.sv:248:29:      Example path: ariane_testharness.i_ariane.i_cva6.icache_dreq_if_cache
                    /home/umerimran/Music/cva6/core/cache_subsystem/cva6_icache.sv:154:3:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/cva6.sv:249:29:      Example path: ariane_testharness.i_ariane.i_cva6.icache_dreq_cache_if
                    /home/umerimran/Music/cva6/core/frontend/frontend.sv:294:5:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/cva6.sv:248:29:      Example path: ariane_testharness.i_ariane.i_cva6.icache_dreq_if_cache
```

### Description of Done

Lint Warning should be removed.

umerimran-10xe, 2023-04-03:
Further warnings related to this issue is

2
```
%Warning-UNOPTFLAT: /home/umerimran/Music/cva6/core/issue_stage.sv:108:32: Signal unoptimizable: Circular combinational logic: 'ariane_testharness.i_ariane.i_cva6.issue_stage_i.issue_ack_sb_rename'
  108 |     logic                      issue_ack_sb_rename;
      |                                ^~~~~~~~~~~~~~~~~~~
                    /home/umerimran/Music/cva6/core/issue_stage.sv:108:32:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.issue_ack_sb_rename
                    /home/umerimran/Music/cva6/core/re_name.sv:38:5:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/issue_stage.sv:106:32:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.issue_instr_rename_sb
                    /home/umerimran/Music/cva6/core/scoreboard.sv:118:3:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/issue_stage.sv:110:32:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.issue_instr_sb_iro
                    /home/umerimran/Music/cva6/core/scoreboard.sv:306:27:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/scoreboard.sv:300:38:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.i_scoreboard.rs1_fwd_req
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:302:44:      Example path: ASSIGNW
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:133:33:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.i_scoreboard.i_sel_rs1.gen_arbiter.data_nodes
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:302:44:      Example path: ASSIGNW
```

3
```
%Warning-UNOPTFLAT: /home/umerimran/Music/cva6/core/id_stage.sv:51:38: Signal unoptimizable: Circular combinational logic: 'ariane_testharness.i_ariane.i_cva6.id_stage_i.decoded_instruction'
   51 |     ariane_pkg::scoreboard_entry_t   decoded_instruction;
      |                                      ^~~~~~~~~~~~~~~~~~~
                    /home/umerimran/Music/cva6/core/id_stage.sv:51:38:      Example path: ariane_testharness.i_ariane.i_cva6.id_stage_i.decoded_instruction
                    /home/umerimran/Music/cva6/core/decoder.sv:1188:34:      Example path: ASSIGNW
                    /home/umerimran/Music/cva6/core/id_stage.sv:51:38:      Example path: ariane_testharness.i_ariane.i_cva6.id_stage_i.decoded_instruction
```

4
```
%Warning-UNOPTFLAT: /home/umerimran/Music/cva6/core/issue_stage.sv:94:32: Signal unoptimizable: Circular combinational logic: 'ariane_testharness.i_ariane.i_cva6.issue_stage_i.rs1_iro_sb'
   94 |     logic [REG_ADDR_SIZE-1:0]  rs1_iro_sb;
      |                                ^~~~~~~~~~
                    /home/umerimran/Music/cva6/core/issue_stage.sv:94:32:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.rs1_iro_sb
                    /home/umerimran/Music/cva6/core/scoreboard.sv:306:27:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/scoreboard.sv:300:38:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.i_scoreboard.rs1_fwd_req
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:302:44:      Example path: ASSIGNW
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:133:33:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.i_scoreboard.i_sel_rs1.gen_arbiter.data_nodes
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:302:44:      Example path: ASSIGNW

```

5

```
%Warning-UNOPTFLAT: /home/umerimran/Music/cva6/core/issue_stage.sv:98:32: Signal unoptimizable: Circular combinational logic: 'ariane_testharness.i_ariane.i_cva6.issue_stage_i.rs2_iro_sb'
   98 |     logic [REG_ADDR_SIZE-1:0]  rs2_iro_sb;
      |                                ^~~~~~~~~~
                    /home/umerimran/Music/cva6/core/issue_stage.sv:98:32:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.rs2_iro_sb
                    /home/umerimran/Music/cva6/core/scoreboard.sv:307:27:      Example path: ALWAYS
                    /home/umerimran/Music/cva6/core/scoreboard.sv:300:51:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.i_scoreboard.rs2_fwd_req
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:302:44:      Example path: ASSIGNW
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:133:33:      Example path: ariane_testharness.i_ariane.i_cva6.issue_stage_i.i_scoreboard.i_sel_rs2.gen_arbiter.data_nodes
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:302:44:      Example path: ASSIGNW

```

6

```
%Warning-UNOPTFLAT: /home/umerimran/Music/cva6/core/cache_subsystem/wt_dcache_ctrl.sv:25:43: Signal unoptimizable: Circular combinational logic: 'ariane_testharness.i_ariane.i_cva6.genblk3.i_cache_subsystem.i_wt_dcache.__Vcellout__gen_rd_ports[1].i_wt_dcache_ctrl__req_port_o'
   25 |   output dcache_req_o_t                   req_port_o,
      |                                           ^~~~~~~~~~
[... cut ...]
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:293:54:      Example path: ASSIGNW
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:135:33:      Example path: ariane_testharness.i_ariane.i_cva6.genblk3.i_cache_subsystem.i_wt_dcache.i_wt_dcache_mem.i_rr_arb_tree.gen_arbiter.req_nodes
                    /home/umerimran/Music/cva6/vendor/pulp-platform/common_cells/src/rr_arb_tree.sv:293:54:      Example path: ASSIGNW
```

umerimran-10xe, 2023-04-03:
There are many others as well in mmu, we will resolve these first and then go ahead with the other ones in `core/` directory.

JeanRochCoulon, 2024-05-22:
This task is maybe completed, @umerimran-10xe do you confirm ?  In that case, the task could be closed.

closed 2024-05-22 by fatimasaleem