openhwfoundation/cva6 #1161: MIP CSR is failing while verifying access mode

spidugu444, 2023-04-03:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

While verifying Access mode for MIP(Machine Interrupt Pending) CSR on CVA6 core, test got failed on RTL end due to the default reset value as it is reading as 0x00000080.
As per riscv priviliged specification default reset value for MIP CSR is 0x00000000. 

For quick reference please find attached log files.
[riscv_vcs_mip_csr_test_0.log](https://github.com/openhwgroup/cva6/files/11136877/riscv_vcs_mip_csr_test_0.log)
[riscv_spike_mip_csr_test_0.log](https://github.com/openhwgroup/cva6/files/11136878/riscv_spike_mip_csr_test_0.log)

Note: on spike end test is working fine.

MikeOpenHWGroup, 2023-08-23:
Hi @frikhaAziz, can you have a look at this one?

MikeOpenHWGroup, 2023-08-23:
Hi @ASintzoff.  @JeanRochCoulon let me know that @frikhaAziz has completed his internship, so I am assigning this one to you.

ASintzoff, 2023-11-13:
CSR mip: 0x00000080 => bit 7 (mip.MTIP) is set

JeanRochCoulon, 2025-01-02:
@zchamski We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case ?

zchamski, 2025-01-09:
Solved in https://github.com/openhwgroup/cva6/issues/1974, can be closed.

closed 2025-01-09 by JeanRochCoulon