openhwfoundation/cva6 #1162: In-line delays in SimDTM.sv

MikeOpenHWGroup, 2023-04-03:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

There are a number of "pound delays" in `corev_apu/tb/common/SimDTM.sv` ([example](https://github.com/openhwgroup/cva6/blob/6d006558f196ce21dec73d4f926f7ae5d67398e0/corev_apu/tb/common/SimDTM.sv#L38)).

Is there a reason for these to exist?

pcotret, 2023-10-19:
Recent upgrade to Verilator ` v5.008`, got these warnings:
```bash
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimJTAG.sv:43:17: Use --timing or --no-timing to specify how delays should be handled
                                                                                      : ... In instance ariane_testharness.i_SimJTAG
   43 |    wire         #0.1 __jtag_TDO = jtag_TDO_driven ?
      |                 ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimJTAG.sv:54:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                      : ... In instance ariane_testharness.i_SimJTAG
   54 |    assign #0.1 jtag_TCK   = __jtag_TCK;
      |           ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimJTAG.sv:55:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                      : ... In instance ariane_testharness.i_SimJTAG
   55 |    assign #0.1 jtag_TMS   = __jtag_TMS;
      |           ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimJTAG.sv:56:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                      : ... In instance ariane_testharness.i_SimJTAG
   56 |    assign #0.1 jtag_TDI   = __jtag_TDI;
      |           ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimJTAG.sv:57:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                      : ... In instance ariane_testharness.i_SimJTAG
   57 |    assign #0.1 jtag_TRSTn = __jtag_TRSTn;
      |           ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimJTAG.sv:59:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                      : ... In instance ariane_testharness.i_SimJTAG
   59 |    assign #0.1 exit = __exit;
      |           ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:38:8: Use --timing or --no-timing to specify how delays should be handled
                                                                                    : ... In instance ariane_testharness.genblk2.i_SimDTM
   38 |   wire #0.1 __debug_req_ready = debug_req_ready;
      |        ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:39:8: Use --timing or --no-timing to specify how delays should be handled
                                                                                    : ... In instance ariane_testharness.genblk2.i_SimDTM
   39 |   wire #0.1 __debug_resp_valid = debug_resp_valid;
      |        ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:40:15: Use --timing or --no-timing to specify how delays should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   40 |   wire [31:0] #0.1 __debug_resp_bits_resp = {30'b0, debug_resp_bits_resp};
      |               ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:41:15: Use --timing or --no-timing to specify how delays should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   41 |   wire [31:0] #0.1 __debug_resp_bits_data = debug_resp_bits_data;
      |               ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:50:10: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   50 |   assign #0.1 debug_req_valid = __debug_req_valid;
      |          ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:51:10: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   51 |   assign #0.1 debug_req_bits_addr = __debug_req_bits_addr[6:0];
      |          ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:52:10: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   52 |   assign #0.1 debug_req_bits_op = __debug_req_bits_op[1:0];
      |          ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:53:10: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   53 |   assign #0.1 debug_req_bits_data = __debug_req_bits_data[31:0];
      |          ^
[... cut ...]
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   54 |   assign #0.1 debug_resp_ready = __debug_resp_ready;
      |          ^
%Error-NEEDTIMINGOPT: /home/pascal/Documents/cva6/corev_apu/tb/common/SimDTM.sv:55:10: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                     : ... In instance ariane_testharness.genblk2.i_SimDTM
   55 |   assign #0.1 exit = __exit;
      |          ^

````

fatimasaleem, 2023-10-19:
@pcotret you will need to use `--no-timing` flag to avoid this.

pcotret, 2023-10-19:
I tried, it solves my issue. Thanks @fatimasaleem

closed 2023-10-19 by JeanRochCoulon