openhwfoundation/cva6 #1171: Spike Configuration for CVA6 CSR Verification

spidugu444, 2023-04-05:
Hi,
 while verifying access modes for PMPADDR CSR I am facing issues on spike configuration as mentioned below.

As CVA6 core has a granularity of 8 bytes that bit-0 of the PMPADDRCSR has been hardcoded to 0, After making bit-0 to zero on RTL end test is running fine but on spike end default reset value of PMPADDR0 CSR is reading as 0xFFFFFFFF.

for PMPADDR1-PMPADDR7 CSRs after making bit-0 to zero, on RTL end test is working fine, so to make test working on spike end we need to configure PMPADDR CSRs bit-0 to zero.

could anyone please help me in configuring spike to resolve above issues.

Thanks,
Sai Krishna

ASintzoff, 2023-10-25:
Related to task https://github.com/openhwgroup/cva6/issues/1403

zchamski, 2024-04-17:
The reset value of `PMPADDR0` can now be set in Spike using the parameter `/top/cores/pmpaddr0` (or its per-core version `/top/core/.../pmpaddr0`).  Testing has yet to be done.

zchamski, 2024-04-17:
However, the root cause of the `0xffffffff` value appears to be the mismatch of PMP granularity (8 in RTL, 4 in Spike unless overridden in the device tree), see https://github.com/openhwgroup/cva6/issues/1403#issuecomment-1851969714.

MarioOpenHWGroup, 2024-06-14:
I think this issue is solved as the core was hardcoding the LSB to 0 even if the region was in OFF. This was solved as can be seen in the RTL https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv#L796

zchamski, 2024-08-26:
fixed, should be closed.

valentinThomazic, 2024-08-26:
Fixed, closing the issue.

closed 2024-08-26 by valentinThomazic