openhwfoundation/cva6 #1176: [TASK] LINT: D - Big bit endian vector: left < right of bit range

umerimran-10xe, 2023-04-10:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

**Environment**

Operating System: [Ubuntu 20.04]
Compiler Version: [gcc 9.4.0]
CVA6 Version: [SHA:7f4549f2]

When compiling CVA6 with Verilator 5.008, a lint error is being generated due to the Big endian format being used in many of the FPnew submodules. It warns that a packed vector is declared with big endian bit numbering (i.e. [0:7]). Little endian bit numbering is now the standard and we should try to fix this, for now suppressing this warning in verilator configuration file and filing this issue for posterity.

```
%Warning-LITENDIAN: /home/umerimran/Music/cva6/vendor/pulp-platform/fpnew/src/fpnew_pkg.sv:57:17: Big bit endian vector: left < right of bit range: [0:4]
                                                                                                : ... In instance ariane_testharness
   57 |   typedef logic [0:NUM_FP_FORMATS-1]       fmt_logic_t;     
      |                 ^
```

### Description of Done

Lint Warning should be waived or removed.

umerimran-10xe, 2023-04-10:
2nd case of this warning 

```
%Warning-LITENDIAN: /home/umerimran/Music/cva6/corev_apu/fpga/src/apb_timer/timer.sv:47:11: Big bit endian vector: left < right of bit range: [0:2]
                                                                                          : ... In instance ariane_testharness.i_ariane_peripherals.gen_timer.i_timer.TIMER_GEN[0].timer_i
   47 |     logic [0:'d2] [31:0]  regs_q, regs_n;
      |           ^
```

umerimran-10xe, 2023-05-05:
Closing this issue since it is resolved by #1102

umerimran-10xe, 2023-05-05:
This one still remains, I had turned it off earlier in the verilator config.

umerimran-10xe, 2023-05-05:
A bit changed to this now.

```
%Warning-LITENDIAN: /home/umerimran/Music/cva6/vendor/openhwgroup/cvfpu/src/fpnew_pkg.sv:50:28: Big bit endian vector: left < right of bit range: [0:4]
                                                                                              : ... In instance ariane_testharness
   50 |   localparam fp_encoding_t [0:NUM_FP_FORMATS-1] FP_ENCODINGS  = '{
      |                            ^
```

fatimasaleem, 2023-09-08:
Closing this, since this is in vendor directory.

closed 2023-09-08 by fatimasaleem