openhwfoundation/cva6 #1178: PMP CSRs are failing while verifying access modes

spidugu444, 2023-04-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

While verifying access modes for PMPADDR[8]- PMPADDR[15] CSRs are reading zero even after writing it with non-zero value.

while verifying access modes for PMPCFG CSRs PMPCFG[0]- PMPCFG[1] CSRs while writing any random value it is always reading the value that we write after reset.

For PMPCFG[2] - PMPCFG[3] CSRs are reading zero even after writing it with non-zero value.

please let me know if any additional information needed.

Thanks,

MikeOpenHWGroup, 2023-04-11:
Hi @spidugu444, thanks for creating this issue.  I took the liberty of editing it because you are reporting three issues here.

I _believe_ (do not know for sure) that issues 1 and 3 are documentation issues and not an RTL problem.  By inspection, it seems that for the CV32A60X, the number of PMP Entries is 8.  That would imply that PMPADDR[8..15] and PMPCFG[2..3] _do not exist_.

If that is true, then we need to update the [CV32A6 Control Status Registers](https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CV32A6_Control_Status_Registers.rst) chapter in the [CVA6 User Manual](https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/index.html) to indicate:
- That only PMPADDR[0..7] and PMPCFG[0..1] exist.
- What is the correct behaviour if PMPADDR[8..15] and PMPCFG[2..3] are read.

@JeanRochCoulon or @ASintzoff can you comment on this?

Issue 2 is more interesting.  @spidugu444 do you have a test-program we can use to reproduce this?

spidugu444, 2023-04-13:
Hi @MikeOpenHWGroup, please find the steps to replicate PMPCFG0 issue as mentioned below.

1.git clone https://github.com/spidugu444/core-v-verif.git
2.git checkout pmpcfg0_issue
3. Command to run PMPCFG0 test **source cva6/regress/smoke-tests.sh**
4. Test file path **core-v-verif/cva6/tests/custom/csr_access_tests/riscv_pmpcfg0_csr_test_0.S**
5. Yaml file path **core-v-verif/cva6/tests/custom/csr_access_yaml/cva6_pmpcfg0_csr_test.yaml**

MikeOpenHWGroup, 2023-04-13:
Thanks for the additional information @spidugu444.

@ASintzoff and @JeanRochCoulon, I have assigned this issue to you.  I believe this issue should be split into two issues:
1. **Documentation**: what is the correct behaviour if PMPADDR[8..15] and PMPCFG[2..3] are read?
2. **RTL**: why do PMPCFG[2] - PMPCFG[3] read zero even after writing it with non-zero value?

What do you think?

spidugu444, 2023-07-17:
Regarding CVA6 PMPCFG CSRs, As per riscv priviliged specification 

[image]

[image]

The lock bit is enabled by the riscv-dv generated test because of that, in PMPCFG[0]- PMPCFG[1],  while writing any random value I am reading the lock bit enable value. this values will be locked untill the hart is reset. 

With manual test, I am reading expected values on RTL end, but there is a spike dependence problem.

spidugu444, 2023-07-17:
@RanjanThales

spidugu444, 2023-07-18:
@MikeOpenHWGroup

Moschn, 2023-07-18:
Hi @spidugu444,

I do not understand the issues that you observe. 

Why should PMPCFG[0-1] not read the value that were written? I also do not get the locked issue. Could you please provide a sequence of instructions that trigger the behavior? 

The other two issues mentioned in the initial report are documentation issues. @MikeOpenHWGroup is correct: only 8 PMP entries are enabled in the default configuration. Therefore, any write to PMPADDR[8-15] or PMPCFG[2-3] will not have any effect, and any read will read 0s.

MikeOpenHWGroup, 2023-08-23:
Hi @spidugu444, can you respond to @Moschn's question?

zchamski, 2024-04-17:
Further investigation of point 2 is needed as the order of writes into the `PMPCFGn` registers impacts future operation of the registers (`L` bit, reserved combination `R=0, W=1`).  In particular, it will be interesting to analyze the CSR description of `PMPCFGn` properties as captured by `riscv-config` (https://github.com/riscv-software-src/riscv-config, vendorized in CVA6 as of #2045).

JeanRochCoulon, 2025-01-02:
@zchamski We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case?

zchamski, 2025-01-09:
@JeanRochCoulon: Since the PMP will not be used in CV32A65X, please change the label to "not CV32A65X".