openhwfoundation/cva6 #1181: [BUG] replace FORMAL directive by VERILATOR

JeanRochCoulon, 2023-04-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

In PMP modules, some  " `ifdef FORMAL " are inserted in the code to enable/disable assertions.
In other cva6 modules, assertions are enabled by " `ifndef VERILATOR ". Can we replace FORMAL by VERILATOR ? The goal is to uniformize the define directives to bettre drive and use them.

JeanRochCoulon, 2023-04-12:
@Moschn

Moschn, 2023-04-13:
The FORMAL ifdef surrounds any kind of assert that was at some point checked by a formal model checker. I am not sure if verilator even compiles with FORMAL defined. I guess one option is to just remove all asserts, as no one has used the formal tool in the last few years.

JeanRochCoulon, 2025-01-09:
FORMAL does not exist anymore in RTL code. Close the issue.

closed 2025-01-09 by JeanRochCoulon