openhwfoundation/cva6 #1202: CVA6 custom CSRs Address Configuration issue

spidugu444, 2023-04-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

While verifying access modes for CVA6 CSRs, I found an issue on custom CSRs(ICACHE and DCACHE) address configuration as mentioned below.

As per riscv priviliged specification address for RW custom CSRs will be in range of 0x7C0-0x7FF or 0xBC0-0xBFF, but in CVA6 core address for custom CSRs are in range of **0x700-0x77F** which is for Standard read/write.

please find below snaps for better understand.

CVA6 Custom CSRs address range as per [CVA6 user manual](https://github.com/openhwgroup/cva6/issues/new?assignees=&labels=Type%3ABug&template=bug.yaml&title=%5BBUG%5D+%3Ctitle%3E) is as shown below.

[image]

As per riscv priviliged specification address range will be as shown below.
[image]

Thanks,

spidugu444, 2023-04-20:
@RanjanThales

frikhaAziz, 2023-07-03:
I pushed a commit to solve this issue. Please review it and close it if there is no problem. @spidugu444

spidugu444, 2023-07-03:
sure @frikhaAziz, I will let you know.

spidugu444, 2023-07-05:
@frikhaAziz, good day. Your RTL tweaks have allowed me to access the ICACHE and DCACHE CSRs at a custom read/write address range at 0x7C0-0x7C1.

spidugu444, 2023-07-05:
@JeanRochCoulon, On the spike end, the custom CSRs ICACHE and DCACHE have not been implemnted.

MikeOpenHWGroup, 2023-08-23:
Hi @spidugu444, it seems that the original problem reported in this issue has been solved by @frikhaAziz.  If you agree please:
- close this issue.
- open a new issue against the Spike model to get the custom CSRs implemented.

spidugu444, 2023-08-28:
Closing this issue as I raised a new [issue](https://github.com/openhwgroup/cva6/issues/1346) for CVA6 CSRs spike discrepancies.

closed 2023-08-28 by spidugu444