openhwfoundation/cva6 #1210: [TASK] CI: Test further CVA6 configurations in GitHub CI

niwis, 2023-04-26:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Currently, the GitHub CI runs regression tests only for the default CVA6 configuration `cv64a6_imafdc_sv39`. It is desirable to add further configurations to the tests, e.g. 32-bit CVA6 or CVA6 with the write-back cache. This can be easily done using a Matrix:

```patch
diff --git a/.github/workflows/ci.yml b/.github/workflows/ci.yml
index 052d7736..836abecf 100644
--- a/.github/workflows/ci.yml
+++ b/.github/workflows/ci.yml
@@ -13,6 +13,7 @@ jobs:
     strategy:
       matrix:
         testcase: [asm-tests, mul, amo, fp, benchmarks]
+        target: [cv64a6_imafdc_sv39, cv32a60x, cv32a6_imafc_sv32]
     env:
       RISCV: /riscv
     steps:
@@ -22,4 +23,4 @@ jobs:
     - name: Prepare
       run: ci/setup.sh
     - name: run tests
-      run: make run-${{ matrix.testcase}}-verilator
+      run: make run-${{ matrix.testcase}}-verilator target=${{ matrix.target }}
```

The 32-bit tests currently fail, though, and need to be fixed before this can be done.

### Description of Done

GitHub CI tests multiple targets.

JeanRochCoulon, 2024-05-22:
HEllo @niwis 
Some PRs have been submitted and merged in that way, what is the status of this task ?   If you think it is not completed, what are the next steps ?

niwis, 2024-05-22:
Yes, this was done in #1523 and #2114.

closed 2024-05-22 by niwis