openhwfoundation/cva6 #1218: [BUG] Questasim Flow broken

christian-lanius, 2023-05-02:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Since the changes in #1125 the questasim based simulations time out: There is no mechanism with the default Makefile to signal and end of simulation, thus the simulation times out. The following changes have to be made:

- Add +tohost_addr=80001000 to the questa-flags or questa-cmd variable. As far as I can tell, this should always be done, in order to be aligned with the test suite. This value being undefined is also the reason why there are warnings reported in the CI job (which uses verilator)
- Add RVFI_PORT back to the defines in order to get the RVFI to actually trigger the end of simulation

To reproduce the bug: 

1. Download the latest master branch (4f06aa620f75bcae369f05d0652283d45ef76a24 at the time of writing)
2. Set up the riscv ISA and benchmark suite
3. Try to run any example program in questasim, which uses the tohost address to stop the simulation: `make sim elf-bin=tmp/riscv-tests/build/isa/rv64ui-v-lw batch-mode=1`
4. Observe the simulation failing: The to host value is 0xFFFFFFFF and the simulation only stops because of the timeout being reached

The expected behavior would be that the simulation finishes with tohost being set to 0.

christian-lanius, 2023-05-02:
Looking at it some more, with the "default" way of compiling the test suite (running the [ci/build-riscv-tests.sh](https://github.com/openhwgroup/cva6/blob/master/ci/build-riscv-tests.sh) script), the "tohost_addr" changes and is not always 0x80001000. This leads to some tests not passing (for example `rv64uc-p-rvc` and `rv64ud-p-move`). So just hardcoding it to that value in the Makefile is wrong.

JeanRochCoulon, 2023-10-24:
Related to #1569