openhwfoundation/cva6 #1220: [BUG] Performance Discrepancy between verilator and questasim for benchmarks

christian-lanius, 2023-05-02:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I am unsure if this is a bug, or if this is somehow related to preloading etc.
Looking at the CI runs (and validating by running verilator myself), some benchmarks perform significantly better than what I observe running them in questasim (First row is mcycle, second is minstret).:

| Benchmark  | Verilator | Questasim |
| ------------- | --------- | --------- |
| Dhrystone (mcycle )    | 246878   | **750468** |
| Dhrystone (minstret)   | 196529   | 196029 |
| Median (mcycle )        | 7692      | 7813 |
| Median (minstret)       | 4159       | 4156 |
| Multiply (mcycle )       | 33482  | 33991|
| Multiply (minstret)      | 24104 | 24101|
| QSort (mcycle )       | 230134 | 270068 |
| QSort (minstret)      | 123803 | 127368 |
| RSort (mcycle )       | 224664 | **813692** |
| RSort (minstret)      | 171157 | 171154|
| towers (mcycle )       | 6815 | **46534** |
| towers (minstret)      | 6170 | 4855 |
| vvadd (mcycle )       | 2462 | 4700 |
| vvadd  (minstret)      | 2420 | 2417 |

I have used the command ` make run-benchmarks batch-mode=1 max_cycles=10000000`. Is this is the expected behavior, i.e. that the benchmarks do not produce the same results? I have tried to make them align, but I did not find a way to do so. This currently "looks like" the IPC are sometimes orders of magnitude worse when simulated in questasim.

How to reproduce:

1. Run `make sim elf-bin=tmp/riscv-tests/build/benchmarks/towers.riscv batch-mode=1` and note the reported mcycle and minstret values
2. Run `work-ver/Variane_testharness tmp/riscv-tests/build/benchmarks/towers.riscv` and note the reported mcycle and minstret values

Expected behavior: The values should be roughly identical. 
Observed behavior: The value for the number of cycles differs drastically.

christian-lanius, 2023-05-02:
Additionally, two tests from the test-suite fail in questasim, but not in verilator.

`rv64uc-p-rvc` test fails in questasim, but simulates properly in verilator. It passes in verilator after 35843 cycles, but will enter an infinite loop in questasim. I ran it with the command ` make sim elf-bin=tmp/riscv-tests/build/isa/rv64uc-p-rvc batch-mode=1`. After 6933µs an exception with cause "Environment Call User Mode" is triggered. Afterwards, the processor gets stuck in an infinite loop:
```
 6935970ns   346791 M 000000008000003c 0 00003f17 auipc          t5, 0x3               t5  :000000008000303c
 6936010ns   346793 M 0000000080000040 0 fc3f2223 sw             gp, -60(t5)           gp  :0000000000000001 t5  :000000008000303c VA: 0000000080003000 PA: 00000080003000
 6936010ns   346793 M 0000000080000044 0 ff9ff06f j              pc - 8    
```

`rv64ud-p-move` passes in verilator with tohost=0. In questasim, it instead results in a return code of 143165915, but it does finish (in contrast to the above).

JeanRochCoulon, 2023-05-02:
@cathales has studied Dhrystone. Can you cross check the figures with @christian-lanius ?

cathales, 2023-05-03:
Hi,

It seems that I am not using the same options to build Dhrystone as I get 221425 cycles for 195508 instructions. It is consistent with your Verilator results.

I use core-v-verif to run simulations so I don't exactly know what cva6 scripts & makefiles do. Below are a few remarks which I hope will help.

TL;DR: binaries might differ, but there seems to be RTL differences not related to configuration alone.

The two simulators do not provide the exact same number of executed instructions so either you do not provide the exact same binary or you simulate different cores where at least one of them has bugs. (Or both :wink:)

Note that IPC is significantly different between the two simulators (0.796 vs. 0.261 for Dhrystone) so is must not be only about compiler flags : the RTL must be different. One instruction every 4 cycles in average (0.261) is really low. Maybe it is about hardware configuration? With the different configurations I have tried I never found such a ×3 factor in performance so I also do not think it is only a matter of hardware configuration.

JeanRochCoulon, 2023-05-12:
@cathales @christian-lanius    An issue not assigned is a dead issue. I re-assigned it to both of you !!